realsenseai/realsense-ros #1003: SR305: Incomplete bulk usb transfer!

ruoxueruoxi90915, 2019-11-20:
1. roslaunch realsense2_camera rs_rgbd.launch: [ERROR] [1574219710.066283358]: Incomplete bulk usb transfer!
2 . When I use a 3-meter USB 3.0 extension, I run roslaunch realsense2_camera rs_rgbd.launch: [ WARN] [1574219692.641234330]: No RealSense devices were found!
3. Can this realsense-ros package support both D435i and SR300? (the version of the RealSense SDK is 2.29.0)

doronhi, 2019-11-27:
1. Is the camera available using realsense-viewer?
2. Yes. This realsense2_camera wrapper supports both SR300 and D435i.

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

ruoxueruoxi90915,

Are you still having issues with item 1) below? We are able to run rs_rgbd.launch successfully.

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-04-05 by RealSenseCustomerSupport