realsenseai/realsense-ros #1005: Release realsense2_description into melodic

peci1, 2019-11-22:
Hi, realsense2_description is a simple package with almost no dependencies. Could you please release at least this one into melodic? It shouldn't be a problem. And it would help us - We've got visualization (robot model) packages separated from the driver part, and currently, to get realsense URDF description, we'd need to clone and build the whole realsense2 repo, which unfortunately also brings in the realsense2_camera package, which we might not want to build just for the visualization clients. And our "automation" (a.k.a. wstool) doesn't support cloning a repo and building only some packages from it.

Thanks for considering this request.

eisenWu, 2019-12-10:
Hi,
i am using JETSON TX2 with melodic, when i launch the T265 these are aways some errors 
```
[[ERROR] [1575669433.435019328]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1575669433.435131872]: The error before refreshing the cache was: Could not find library corresponding to plugin realsense2_camera/RealSenseNodeFactory. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[FATAL] [1575669433.435440256]: Failed to load nodelet '/camera/realsense2_camera` of type `realsense2_camera/RealSenseNodeFactory` to manager `realsense2_camera_manager'
[camera/realsense2_camera-2] process has died [pid 25692, exit code 255, cmd /opt/ros/melodic/lib/nodelet/nodelet load realsense2_camera/RealSenseNodeFactory realsense2_camera_manager __name:=realsense2_camera __log:=/home/robocar/.ros/log/7a58a8f4-1863-11ea-ac4d-00044bc5026c/camera-realsense2_camera-2.log].
log file: /home/robocar/.ros/log/7a58a8f4-1863-11ea-ac4d-00044bc5026c/camera-realsense2_camera-2*.
```
is there some ploblems with melodic?

RealSenseCustomerSupport, 2020-01-14:
----------------------------------------------

Hi @peci1,

You can build the whole wrapper for Melodic instead of having realsense2_description
only.Please download the latest ROS Wrapper 2.0 (build 2.2.11) and try to build and run it with ROS Melodic.You might need to install ros-melodic-ddynamic-reconfigure :
sudo apt-get install ros-melodic-ddynamic-reconfigure

Please let us know if you need any further assistance with this matter.

@eisenWu,

Please open the separate thread for your question as it is not related to the question we confer about in current thread.

Thank you,
David

peci1, 2020-01-14:
Well, of course I can clone and build the package. Just wanted to say that it's easier to install via APT, and since you were already able to cut out ddynamic_reconfigure, I thought realsense2_description would be another meaningful step towards full-binary melodic. Common, melodic's been here for almost 2 years and you don't yet provide official binary packages for it...

RealSenseCustomerSupport, 2020-01-24:
----------------------------------------------

 @peci1,

Please check back often for official binary packages availability. Here is the link to submitted PR :
https://github.com/ros/rosdistro/pull/23493

RealSenseCustomerSupport, 2020-01-24:
----------------------------------------------

@peci1,

Please update us after you have the package installed and let us know if you need any additional help on this issue.

mcamurri, 2020-02-12:
> Hi @peci1,
> 
> You can build the whole wrapper for Melodic instead of having realsense2_description
> only.Please download the latest ROS Wrapper 2.0 (build 2.2.11) and try to build and run it with ROS Melodic.**You might need to install ros-melodic-ddynamic-reconfigure :**
> sudo apt-get install ros-melodic-ddynamic-reconfigure
> 
> Please let us know if you need any further assistance with this matter.
> 
> 
> Thank you,
> David

If I try to compile the latest `development` branch (commit 69d199d9e01a7c57b78831a6486c83ad8653aba9) and install `ros-melodic-ddynamic-reconfigure` from apt-get, I get the following error:
```
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::readAndSetDynamicParam(ros::NodeHandle&, std::shared_ptr<ddynamic_reconfigure::DDynamicReconfigure>, std::__cxx11::string, int, int, rs2::sensor, int*)’:
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:346:14: error: ‘using element_type = class ddynamic_reconfigure::DDynamicReconfigure {aka class ddynamic_reconfigure::DDynamicReconfigure}’ has no member named ‘registerVariable’
     ddynrec->registerVariable<int>(
              ^~~~~~~~~~~~~~~~
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:346:31: error: expected primary-expression before ‘int’
     ddynrec->registerVariable<int>(
                               ^~~
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:347:135: error: expected primary-expression before ‘,’ token
         option_name, *option_value, [this, sensor, option_name](int new_value){set_auto_exposure_roi(option_name, sensor, new_value);},
                                                                                                                                       ^
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::registerAutoExposureROIOptions(ros::NodeHandle&)’:
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:373:22: error: ‘using element_type = class ddynamic_reconfigure::DDynamicReconfigure {aka class ddynamic_reconfigure::DDynamicReconfigure}’ has no member named ‘publishServicesTopics’
             ddynrec->publishServicesTopics();
                      ^~~~~~~~~~~~~~~~~~~~~
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp: In member function ‘void realsense2_camera::BaseRealSenseNode::registerDynamicOption(ros::NodeHandle&, rs2::options, std::__cxx11::string&)’:
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:403:22: error: ‘using element_type = class ddynamic_reconfigure::DDynamicReconfigure {aka class ddynamic_reconfigure::DDynamicReconfigure}’ has no member named ‘registerVariable’
             ddynrec->registerVariable<bool>(
                      ^~~~~~~~~~~~~~~~
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:403:39: error: expected primary-expression before ‘bool’
             ddynrec->registerVariable<bool>(
                                       ^~~~
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:405:89: error: expected primary-expression before ‘,’ token
               [option, sensor](bool new_value) { sensor.set_option(option, new_value); },
                                                                                         ^
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:431:24: error: ‘using element_type = class ddynamic_reconfigure::DDynamicReconfigure {aka class ddynamic_reconfigure::DDynamicReconfigure}’ has no member named ‘registerVariable’
               ddynrec->registerVariable<int>(
                        ^~~~~~~~~~~~~~~~
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:431:41: error: expected primary-expression before ‘int’
               ddynrec->registerVariable<int>(
                                         ^~~
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:433:92: error: expected primary-expression before ‘,’ token
                   [option, sensor](int new_value) { sensor.set_option(option, new_value); },
                                                                                            ^
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:451:28: error: ‘using element_type = class ddynamic_reconfigure::DDynamicReconfigure {aka class ddynamic_reconfigure::DDynamicReconfigure}’ has no member named ‘registerVariable’
                   ddynrec->registerVariable<double>(
                            ^~~~~~~~~~~~~~~~
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:451:45: error: expected primary-expression before ‘double’
                   ddynrec->registerVariable<double>(
                                             ^~~~~~
/home/mcamurri/catkin_ws/src/realsense/realsense2_camera/src/base_realsense_node.cpp:453:99: error: expected primary-expression before ‘,’ token
                       [option, sensor](double new_value) { sensor.set_option(option, new_value); },
                                                                                                   ^
[... cut ...]
CMakeFiles/Makefile2:144: recipe for target 'CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
```
it seems the API has changed from kinetic to melodic at some point, because the pal-robotics' `kinetic-devel` branch works instead.

The current way to make it compile on melodic is by cloning [the pal-robotic fork](https://github.com/pal-robotics/ddynamic_reconfigure), as mentioned in #689

doronhi, 2020-02-13:
@mcamurri , I think you have got the wrong version of ddynamic-reconfigure because I had to do nothing when switched from kinetic to melodic. In any case, this is a separate issue.

regarding the realsense2_description package for melodic, it is available now (using `sudo apt-get install ros-melodic-realsense2-description`). Can someone confirm that this issue can be closed? @peci1 ?

mcamurri, 2020-02-14:
> @mcamurri , I think you have got the wrong version of ddynamic-reconfigure because I had to do nothing when switched from kinetic to melodic. In any case, this is a separate issue.

Yes you are right. I didn't realize I was using an unofficial mirror of the ROS ppa. With the official ROS mirror it compiles.

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@peci1, @mcamurri,

Did you have a chance to install the realsense2_description package for Melodic? Do you need any further assistance with this matter?

peci1, 2020-02-14:
I won't install it this month since we need reliability for the DARPA SubT challenge, but in March we'll definitely want to switch to the released packages.

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

@peci1,

The realsense2_description package for Melodic was released, so the original issue is solved and will be closed. If you will experience any issues with the package, please open a new thread and we'll be glad to provide you with all the needed help.

Thank you!

closed 2020-02-18 by RealSenseCustomerSupport