realsenseai/realsense-ros #1006: API Mismatch

Rohan035, 2019-11-29:
[ERROR] [1575024018.750131608]: Failed to load nodelet [/camera/realsense2_camera] of type [realsense2_camera/RealSenseNodeFactory] even after refreshing the cache: API version mismatch: librealsense.so was compiled with API version 2.30.0 but the application was compiled with 2.25.0! Make sure correct version of the library is installed (make install)

doronhi, 2019-12-01:
checkout this answer:
https://github.com/IntelRealSense/realsense-ros/issues/950#issuecomment-557136583

RealSenseCustomerSupport, 2020-03-25:
----------------------------------------------

Rohan035....was doronhi's suggestion helpful?

If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-04-05 by RealSenseCustomerSupport