realsenseai/realsense-ros #1015: Unable to enable D435 / D435i on NVIDIA TX2 platform

cclo-astri, 2019-12-10:
Hi all,

Sorry for such long question. I want to use D435 / D435i on NVIDIA TX2 platform, but encountered some difficulties.

Here is the setup procedure that I have done (natively on NVIDIA TX2):

1. Compile librealsense from scratch 
`sudo apt-get install git libssl-dev libusb-1.0-0-dev pkg-config libgtk-3-dev libglfw3-dev`
`cd ~/RealSense/librealsense-master`
`./scripts/setup_udev_rules.sh`
`mkdir build && cd build`
`cmake ../ -DBUILD_EXAMPLES=true -DFORCE_LIBUVC=true -DBUILD_WITH_CUDA=true -DCMAKE_BUILD_TYPE=release`
`make`
`sudo make install`

1. Compuile realsense-ros wrapper
`sudo apt-get install ros-kinetic-ddynamic-reconfigure`
`git clone https://github.com/IntelRealSense/realsense-ros.git`
`cp realsense-ros/realsense2_* <my_catkinspace>/src`
`cd <my_catkinspace>`
`catkin_make <with lots of compile flags used by my SLAM application>`

1. Plug in D435i to TX2

> [85737.656518] usb 2-1: new SuperSpeed USB device number 8 using xhci-tegra
[85737.677076] usb 2-1: feature bit otg_vbus_off set
[85737.681796] usb 2-1: New USB device found, idVendor=8086, idProduct=0b3a
[85737.688512] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[85737.695652] usb 2-1: Product: Intel(R) RealSense(TM) Depth Camera 435i
[85737.702183] usb 2-1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
[85737.709148] usb 2-1: SerialNumber: 903223051641
[85737.714290] xhci-tegra 3530000.xhci: tegra_xhci_mbox_work mailbox command 6
[85737.723844] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[85737.731171] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[85737.738679] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[85737.756139] uvcvideo: Unable to create debugfs 2-8 directory.
[85737.762057] input: Intel(R) RealSense(TM) Depth Ca as /devices/3530000.xhci/usb2/2-1/2-1:1.0/input/input10
[85737.772670] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[85737.779991] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[85737.793222] uvcvideo: Unable to create debugfs 2-8 directory.
[85744.028180] xhci-tegra 3530000.xhci: entering ELPG
[85744.035789] xhci-tegra 3530000.xhci: entering ELPG done

1. Run `roslaunch realsense2_camera rs_camera.launch`, but tons of errors message appears ("......" means snipped as the original log is too long. I also attached the complete log here: [rs.log](https://github.com/IntelRealSense/realsense-ros/files/3943790/rs.log))
> ........
[ INFO] [1575966281.853248614]: Initializing nodelet with 6 worker threads.
[ INFO] [1575966281.932813334]: RealSense ROS v2.2.9
[ INFO] [1575966281.932921558]: Running with LibRealSense v2.30.0
[ INFO] [1575966282.285488736]:
 10/12 08:24:42,776 WARNING [547646685584] (types.cpp:48) hwmon command 0x4f failed. Error type: No data to return (-21).
[ INFO] [1575966282.828864078]: Device with serial number 845112070474 was found.
[ INFO] [1575966282.828943022]: Device with physical ID /dev/bus/usb/002/008 was found.
[ INFO] [1575966282.828985870]: Device with name Intel RealSense D435I was found.
[ WARN] [1575966282.830108657]: Error extracting usb port from device with physical ID: /dev/bus/usb/002/008
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
........
 10/12 08:24:48,453 WARNING [547314725264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ INFO] [1575966288.669097923]: SELECTED BASE:Depth, 0
 10/12 08:24:48,711 WARNING [547314725264] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
[ INFO] [1575966288.788070249]: RealSense Node Is Up!
 10/12 08:24:48,799 WARNING [547314725264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/12 08:24:48,860 WARNING [547314725264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/12 08:24:49,061 WARNING [547314725264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/12 08:24:49,086 ERROR [547172118928] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 10/12 08:24:49,242 WARNING [547314725264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/12 08:24:49,544 WARNING [547314725264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 10/12 08:24:50,086 ERROR [547172118928] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132
 10/12 08:24:50,810 WARNING [547289559440] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 12,
[ WARN] [1575966291.016468357]: Hardware Notification:USB REC overflow,1.57597e+12,Error,Hardware Error
 10/12 08:24:51,311 ERROR [546324869520] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 10/12 08:24:52,858 WARNING [547323113872] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 23,
 10/12 08:24:52,862 WARNING [547323113872] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 12,
[ WARN] [1575966293.047281289]: Hardware Notification:USB REC overflow,1.57597e+12,Error,Hardware Error
 10/12 08:24:54,074 WARNING [547289559440] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 23,
 10/12 08:24:54,642 ERROR [546324869520] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 10/12 08:24:54,906 WARNING [547289559440] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 12,
[ WARN] [1575966295.078637929]: Hardware Notification:USB REC overflow,1.57597e+12,Error,Hardware Error
 10/12 08:24:56,706 WARNING [547323113872] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 23,
 10/12 08:24:56,942 WARNING [547323113872] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 12,
[ WARN] [1575966297.109665027]: Hardware Notification:USB REC overflow,1.57597e+12,Error,Hardware Error
 10/12 08:24:57,918 WARNING [547289559440] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 23,
[ WARN] [1575966298.130295577]: Hardware Notification:IR stream start failure,1.57597e+12,Error,Hardware Error
[ERROR] [1575966298.130412121]: Hardware Reset is needed. use option: initial_reset:=true
 10/12 08:24:58,134 WARNING [547289559440] (uvc-device.cpp:644) interrupt event received: 1, 3, 0, 7, 0, 9,
 10/12 08:24:58,306 ERROR [546324869520] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
 10/12 08:24:58,788 WARNING [547314725264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/12 08:24:58,848 WARNING [547314725264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/12 08:24:58,909 WARNING [547314725264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 10/12 08:24:58,967 WARNING [547172118928] (messenger-libusb.cpp:83) usb_request_queue returned error, endpoint: 132 error: No such device, number: 19
 10/12 08:24:58,967 ERROR [547172118928] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13

Please help. Thanks.

Best regards,
C.C. Lo

doronhi, 2019-12-10:
Could you first check that librealsense example applications work? realsense-viewer, I expect, is not built but you could try out `rs-distance`, `rs-color`? It could give us a lead.

cclo-astri, 2019-12-11:
Hi @doronhi ,

Thanks for your comment.

I just check the D435i with realsense-viewer (fortunately I have built it). There are 3 modules I could switch ON / OFF: Stereo / RGB / Motion. But only RGB fails to work.

The only step that I **HAVE NOT DONE** is patch kernel to build & load patched drivers, which is [https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md](url)

Because I found a recent comment from your colleague to tell us patching Linux kernel is not recommended: [https://github.com/IntelRealSense/librealsense/issues/4969#issuecomment-537834501](url), or I mis-understood his meaning ? 

Do I really need this step to enable D435i on NVIDIA TX2 ?

Thanks.

doronhi, 2019-12-11:
Right now it doesn't sound like a kernel issue but we need more data.
Could you try the following things:
1. running on the NVIDIA machine:
- Post the printout of `rs-enumerate-device` 
- Try running `rs-color`. 
- Try connect the camera using a USB-hub.
2. Try the camera on a different computer.

cclo-astri, 2019-12-11:
Hi @doronhi ,

I used my colleague's NVIDIA TX2 and do some cross-verification, and I found D435i could work on his board perfectly.

After some discussion, we discovered the Linux Kernel version (and related kernel drivers) are slightly different. My board's Linux Kernel and drivers are kept as factory default version (older) but his board is newer because he performed OS / Kernel update accidentally before.

- The 2 boards are both Ubuntu 16.04 Xenial, but the minor version differs.

I amnow trying re-flashing my TX2 board to sync. up the background environment with my colleague's board first.

I will update once I have finished the testing.

Thanks.

doronhi, 2019-12-12:
Sound promising. Looking forward to your update.

cclo-astri, 2019-12-13:
Hi @doronhi 

I finished my test and now D435i works perfectly in NVIDIA TX2.

- realsense-viewer could view all module's frames.
- I could use my SLAM application to subscribe the ROS-topics of D435i and get valid images.

Thanks for your help.

closed 2019-12-13 by cclo-astri

Eslamrady42, 2020-06-21:
can you please tell how did you solve this problem?