realsenseai/realsense-ros #1018: realsense-ros does not work on a Xavier (2.2.10)

mcalafat-earthrover, 2019-12-12:
Hi!

I have installed librealsense 2.30.0 on my Jetson Xavier  with help of this guide [Jetson Installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md).

Then I installed realsense-ros 2.2.10 compatible with the librealsense version installed before.

I also updated the firmware of my D435 depth camera up to date (it is 5.12.01).

My problem is the next: when I try to launch the rs-camera.launch in order to start receiving some visual data, I get this output:
```
NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
    realsense2_camera_manager (nodelet/nodelet)

auto-starting new master
process[master]: started with pid [7130]
ROS_MASTER_URI=http://10.42.0.155:11311

setting /run_id to 5c38206a-1cdd-11ea-909b-00044bcbec78
process[rosout-1]: started with pid [7141]
started core service [/rosout]
process[camera/realsense2_camera_manager-2]: started with pid [7148]
process[camera/realsense2_camera-3]: started with pid [7149]
[ INFO] [1576154768.757544259]: Initializing nodelet with 4 worker threads.
[ INFO] [1576154769.168134666]: RealSense ROS v2.2.10
[ INFO] [1576154769.168619547]: Running with LibRealSense v2.30.0
[ INFO] [1576154769.548749314]:  
[ INFO] [1576154769.931687448]: Device with serial number 832112071551 was found.

[ INFO] [1576154769.931987035]: Device with physical ID /dev/bus/usb/002/002 was found.
[ INFO] [1576154769.932143436]: Device with name Intel RealSense D435 was found.
[ WARN] [1576154769.944870323]: Error extracting usb port from device with physical ID: /dev/bus/usb/002/002
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[ INFO] [1576154769.967489783]: getParameters...
[ INFO] [1576154770.173026266]: setupDevice...
[ INFO] [1576154770.173284483]: JSON file is not provided
[ INFO] [1576154770.173459155]: ROS Node Namespace: camera
[ INFO] [1576154770.173562025]: Device Name: Intel RealSense D435
[ INFO] [1576154770.173640738]: Device Serial No: 832112071551
[ INFO] [1576154770.173797012]: Device physical port: /dev/bus/usb/002/002
[ INFO] [1576154770.173879020]: Device FW version: 05.12.01.00
[ INFO] [1576154770.173962725]: Device Product ID: 0x0B07
[ INFO] [1576154770.174059196]: Enable PointCloud: Off
[ INFO] [1576154770.174138740]: Align Depth: Off
[ INFO] [1576154770.174222253]: Sync Mode: Off
[ INFO] [1576154770.175183829]: Device Sensors: 
[ INFO] [1576154770.176019433]: Stereo Module was found.
[ INFO] [1576154770.177302963]: RGB Camera was found.
[ INFO] [1576154770.177604856]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1576154770.177750410]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1576154770.177871839]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1576154770.177992116]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1576154770.178109962]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1576154770.178262108]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1576154770.179024822]: num_filters: 0
[ INFO] [1576154770.179195878]: Setting Dynamic reconfig parameters.
[ INFO] [1576154775.834670767]: Done Setting Dynamic reconfig parameters.
[ INFO] [1576154775.883505040]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1576154775.885255356]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1576154775.886498666]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1576154775.918985476]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1576154775.920477313]: setupPublishers...
[ INFO] [1576154775.932323185]: Expected frequency for depth = 30.00000
[ INFO] [1576154775.947857709]: Expected frequency for infra1 = 30.00000
[ INFO] [1576154775.957290365]: Expected frequency for infra2 = 30.00000
[ INFO] [1576154775.965605943]: Expected frequency for color = 30.00000
[ INFO] [1576154775.980397636]: setupStreams...
[ INFO] [1576154776.090985022]: insert Depth to Stereo Module
[ INFO] [1576154776.092752720]: insert Color to RGB Camera
[ INFO] [1576154776.094340622]: insert Infrared to Stereo Module
[ INFO] [1576154776.094934633]: insert Infrared to Stereo Module
[ INFO] [1576154776.385708242]: SELECTED BASE:Depth, 0
[ INFO] [1576154776.455847795]: RealSense Node Is Up!
 12/12 12:46:16,501 WARNING [547352297856] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 12/12 12:46:16,563 WARNING [547352297856] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 12/12 12:46:16,646 WARNING [547352297856] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 12/12 12:46:16,707 WARNING [547352297856] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 12/12 12:46:16,791 WARNING [547352297856] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```
In addition, I have tried the camera on my laptop (with librealsense 2.31.0 on it) and it is correctly working and displaying images.

Thank you very much!

mcalafat-earthrover, 2019-12-13:
Finally I have downgraded realsense some versions until I have find a stable version which works.

The final setup is this:

* ROS-realsense:  **2.2.8**
* librealsense: **v2.28.1** (in fact, from v2.25.0 to v2.28.1 are working)
* ROS Melodic on Ubuntu 18.04 LTS.

furushchev, 2019-12-17:
I met the similar case.
I use the same version on Jetson Xavier, and 2 realsense cameras connected on each USB bus.
When I use `realsense-viewer`, I can see color and depth images for both camera (4 viewers on a display), but with `realsense2_camera`, I can see only the images of one camera.

szx0112, 2020-01-30:
@furushchev I face the similar problem when use two or more cameras. It always show device or resource busy even when the cpu usage is below 30%. Did you able to resolve this problem?

furushchev, 2020-01-30:
@szx0112 Hi, I ended up removing librealsense installed via `apt-get` and instead building librealsense from source. After installing built libraries, I do not see the same issue any more, so far. Hope it will help you!

szx0112, 2020-01-30:
@furushchev  Thanks for the quick reply. My librealsense is also build from source with the same sdk and ros version as @mcalafat-earthrover mentioned. On realsense-viewer, it can stream at most 2 cameras with full resolution (color+depth). However when use realsense ros, it always show `rs::backend_error' what(): could not open device. Last Error: device or resource busy` when the 2nd camera is initialized. 

Did you just follow the [JetsonHack](https://github.com/jetsonhacks/installRealSenseSDK) for the sdk installation? Patch kernel? The usb-hub if used is powered (my usb hub is not powered but I don't think it is a major issue)? Sorry for many questions.

furushchev, 2020-01-30:
@szx0112 I followed the script found at https://github.com/jetsonhacks/buildLibrealsense2Xavier but modified to support later L4T.
I tried and checked that it works on the following L4T versions: 
L4T 32.2.1 => https://github.com/jetsonhacks/buildLibrealsense2Xavier/pull/15
L4T 32.3.1 => https://github.com/furushchev/buildLibrealsense2Xavier/tree/32.3.1

szx0112, 2020-01-30:
Great. I am on L4T 32.3.1 (Jetpack 4.3) without kernel patch (as jetsonhack mentioned). I guess it is still needed for ROS? I will try to re-install it with your version. Thank you for the help @furushchev !

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Can we close this ticket if there is no other question?
Thank you.

doronhi, 2020-02-09:
I think we should keep the issue open for now. If indeed building from source, as @furushchev  suggests, fixes the issue then something needs to be fixed the the apt-get version. I am currently looking into issues relating several devices.

furushchev, 2020-02-10:
@doronhi @RealSenseCustomerSupport  I cannot check by myself since I have a few test configuration. Let's wait reports from other users who use multiple devices simultaneously with apt-get configuration.

szx0112, 2020-02-12:
Just add a quick comment here. Finally I end up with the configuration suggested by @furushchev with the downgraded librealsense (build from source). I am now able to run two d435i cameras with full resolution (1280x720) and high frame rate (15HZ) with both color and depth_aligned frames enabled on ROS. If we want to add more cameras, we have to either reduce the resolution or drop the frame rate (actually the frame rate will drop by itself somehow). 

I guess the bottleneck is caused by the usb power supply on xavier. It seems that at most 4.5W can be supplied in total by the two usb ports. However, according to the realsense multi-camera sync whitepaper, each realsense camera may consume more than 2W power (with full resolution).

furushchev, 2020-02-12:
@szx0112 Interesting... FYI, I actually use a PCIe-USB card to support multiple camera devices on Xavier since all USB ports on Xavier developer kit share bandwidth of USB3.0.

https://www.startech.com/Cards-Adapters/USB-3.0/Cards/PCI-Express-USB-3-Card-4-Dedicated-Channels-4-Port~PEXUSB3S44V

szx0112, 2020-02-12:
@furushchev Thanks for sharing this link, we might start to take a look at those PCIe cards. BTW, just curious how many cameras have you tested (with PCIe) on Xavier? Our ideal case is to make 4 or 5 D435i works at full resolutions and frame rate (~30HZ).

RealSenseCustomerSupport, 2020-03-18:
----------------------------------------------

Please let us know if there is any further help needed in regard to the original topic. There's no  activity for 30+ days.
Thank you.

closed 2020-03-24 by RealSenseCustomerSupport