realsenseai/realsense-ros #1020: PROBLEM after lauching : roslaunch realsense2_camera rs_camera.launch

AstroUKN, 2019-12-12:
I don't understand why I get this warning message...

 12/12 15:28:58,774 WARNING [140556978751232] (ds5-timestamp.cpp:64) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1576160938.775043165]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)

Maybe someone here could help me with this issue ! 

Thank you very much

doronhi, 2019-12-15:
Try checking out [this link](https://github.com/IntelRealSense/realsense-ros/issues/932#issuecomment-538720015)

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

@AstroUKN Any other questions with the reply from doronhi? Looking forward to your update. Thanks!

AstroUKN, 2020-01-07:
@RealSenseCustomerSupport I didnt solve my problem but I think that it comes from my Kernel version and the camera installation version dont match. I am working on a kernel 5.0 but it seems to be working with version 4.16 and others but not mine.

Thanks for your update.

doronhi, 2020-01-09:
The issue could be that librealsense2 still does not support kernel 5.0. See [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

@AstroUKN According to the comment of
https://github.com/IntelRealSense/librealsense/issues/4586#issuecomment-583046210 , DKMS package with kernels 5.0 and 5.3 support is available for download. Please try to download and update to see if it can fix your issue. Thanks!

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@@AstroUKN Did you get any chance to try? Looking forward to your update.Thanks!

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

@AstroUKN Could you please update? Thanks!

AstroUKN, 2020-02-25:
Hi, I didn't update yet my problem for lack of time unfortunately...
But a friend of mine downloaded the last version and didn't have any problem. Hopefully it'll work for me !
Thanks for your answers 

I'll try to get back to it as soon as possible !

RealSenseCustomerSupport, 2020-03-11:
----------------------------------------------

@AstroUKN Did you get it through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-17:
----------------------------------------------

@AstroUKN Sorry that we didn't get feedback from you for weeks. Will close this at this point. Please create another new ticket if you still have questions or issues. Thanks!

closed 2020-03-23 by RealSenseCustomerSupport