realsenseai/realsense-ros #1021: No support for ubuntu 18.04 on melodic ?

jediofgever, 2019-12-13:


jediofgever, 2019-12-13:
okay ,Actually I built it without problems, the dependencies are already installed with a full desktop installation of melodic the only additional one that needs to be installed is ddynamic-reconfigure 

which you can get it with ;
`sudo apt-get install ros-melodic-ddynamic-reconfigure`

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

Hi @jediofgever,

We wanted to make sure that you can work without hassle with ROS Melodic. Please let us know if you need any further assistance with this matter.

Thank you!

jediofgever, 2020-01-07:
Hi, 
I did built and run the package but there is some issues I have met Iamnot sure if they are being caused by ROS melodic version 
https://github.com/IntelRealSense/realsense-ros/issues/1047

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

@jediofgever,

It's good to know that build and installation processes went without problems and you were able to run the package.If you don't have additional questions regarding the above mentioned stages, let's close this thread and discuss each topic (that is not related to these stages) in its own thread as a separate issue.

Thank you!

jediofgever, 2020-01-08:
OK @RealSenseCustomerSupport 
Thank you fro your time

closed 2020-01-08 by jediofgever