realsenseai/realsense-ros #1022: D435i IMU time synchronized with depth image

studennis911988, 2019-12-13:
1.  software version
- librealsense : 2.29.0
- realsense-ros : 2.2.11(latest)
- firmware : UVC_5_11_6_250
-  unite_imu_method"        default="copy"

2. Issue
I try to get synchronized IMU & image but when I check the header in rostopic echo, and clearly they are synchronized just the picture showing below. When the timestamp in depth is " 43777373
", there is no correspondance imu data at same timestamp.

[image]

3. My question is 

- by viewing other issue, it seems that current we still can't get  synchronized IMU & image?
- or relating to my realsense couldn't use HID timestamps like the picture showing below?

[image]

MartyG-RealSense, 2019-12-13:
The link below discusses the difficulty of syncing depth and IMU, and suggestions for how to approach it.

https://github.com/IntelRealSense/librealsense/issues/4854

doronhi, 2019-12-15:
As for the "Frame metadata isn't available!" message, checkout [this link](https://github.com/IntelRealSense/realsense-ros/issues/932#issuecomment-538720015)

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

will you be needing further assistance with this issue?

studennis911988, 2020-01-08:
@RealSenseCustomerSupport @doronhi 
There is no conclusion on the issue you mentioned.
Moreover please provide more information on how to get synchronized IMU and image data,  is it caused by not finding HID timestamp or something else?
Thanks!

doronhi, 2020-01-09:
Hi @studennis911988 ,
The link I provided regarding the "Frame metadata isn't available!" message suggests that there might be something wrong with the installation process and specifically with kernel patches. My suggestion is that you trace back the installation instructions you used and see if you might have missed something. Could you provide the method of installation of the kernel patches (debian or code) and your version of OS?
Regarding the IMU-Image sync, I'm not sure I understand your suggestion. 
The images are provided by the camera at, say, 30Hz. The Imu at, say, 400Hz.
Do you suggest a synchronization mode where Imu is transmitted at 30Hz, and interpolated to the last sent image frame?

studennis911988, 2020-01-09:
@doronhi Thanks for your reply.
First of all, my kernel version is
> Linux dennis 4.15.0-72-generic #81~16.04.1-Ubuntu SMP Tue Nov 26 16:34:21 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

I installed librealsense from source code by following the [installation page](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md), and I think I may truly miss to patch the kernel at the time I installed it.
I try to patch the kernel  that's mentioned below, but **things didn't go right**, the metadata error message still exist.
[image]

Regarding sync between image and IMU, let's take your words for example. I wish to get **each image(30HZ) timestamp which is sync with the IMU timestamp**(remain 400HZ), just like the picture showing below.
[image]

Hope you could provide more information ? Thanks in advanced!

doronhi, 2020-02-02:
Sorry for the long delay.
The graph you added suggests that at time (i) there are both an image message and an imu message. Then many IMU messages and again, both Image and IMU message at time (j).
The fact is that the Image and IMU messages are just not synchronized. There is an Image message at time (i) and then an IMU message at i plus a little offset. If you want to have the imu at time (i) you'll have to estimate it by previous and maybe later readings.
The realsense2_camera wrapper has 2 separate callback functions. One is handling the images and the other dealing with the IMU messages. Their timestamps have a shared basis so you can relate them to one another but as to this moment there is no plan on synchronizing them in the way you describe.

closed 2020-06-04 by studennis911988