realsenseai/realsense-ros #1023: Documentation

Valia3, 2019-12-14:
Hello,
Is there any paper explain the theoretical principle used in this code? I also have a problem while using this algorithm with my camera D435i. The imu data it is not synchronized with the RGB images .
Thanks in advance

doronhi, 2019-12-15:
Can you be more specific? What part of the code? What algorithm you are having problems with?
What do you mean by "not synchronized"? How do you test? I'm asking so that we can make sure we understand what you mean, reproduce and confirm the problem.

Valia3, 2019-12-15:
I am talking about SLAM using D435i. I want to know which technique is been using in this implementation and if there any paper about it. I am trying the algorithm on a virtual box hosted by Ubuntu 19.10 and powered by ubuntu 16.04. I meant by that, that the IMU data from the camera is not correct.

doronhi, 2019-12-15:
Are you referring to the example in the wiki [SLAM-with-D435i](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i)?
If that is the case then I am aware that there were problems regarding the imu publishing rate not being fixed. That was corrected in the last release (2.2.11).
Other then that, I suspect there are more issues causing the IMU not to have the expected effect in the rtabmap algorithm, but I don't know what the problems are.
So, if you indeed refer to that example I would appreciate if you could supply more details as to problems with the IMU messages or the launch files in that example.
Or were referring to something else?

RealSenseCustomerSupport, 2020-01-02:
----------------------------------------------

Hi @Valia3

Any update on the questions outlined by doronhi?

RealSenseCustomerSupport, 2020-02-05:
----------------------------------------------

I'll be closing out the thread if there is nothing further necessary.

closed 2020-04-30 by RealSenseCustomerSupport