realsenseai/realsense-ros #1024: Aligned_depth_to_color frame drop (to ~1 HZ) after upgrading firmware from 2.30 to 2.31

szx0112, 2019-12-15:
After upgrading the librealsense firmware to 2.31. I notice the topic /camera/aligned_depth_to_color become really unstable.

I set depth rate as 15HZ.

In 2.30, I can consistently receive 14~15 HZ messages when call rostopic hz /camera/aligned_depth_to_color/image_raw

After upgrading to 2.31, I call the same topic but often receive "no new message" or topic between 1~6 HZ. It is very hard to receive the 15 HZ anymore.

I tried it both on a PC workstation and a NUC using two different D435i cameras. I believe it is not the problem frin the computer and the camera side.

Any comment is appreciate

doronhi, 2019-12-16:
As for now, I am not aware of any performance difference between the versions. Will keep watching.
In the meanwhile, did you compile or installed the debian? If compiled, could you check the cmake flags? Specifically if "CMAKE_BUILD_TYPE=release"?

szx0112, 2019-12-17:
> As for now, I am not aware of any performance difference between the versions. Will keep watching.
> In the meanwhile, did you compile or installed the debian? If compiled, could you check the cmake flags? Specifically if "CMAKE_BUILD_TYPE=release"?

Thanks for the reply. I build everything from the source and cmake with the "CMAKE_BUILD_TYPE=release" flag enabled. After re-install everything, the PC works good now, but the NUC still show similar results. rostoptic hz /camera/aligned_depth_to_color/image_raw remain low frequency (1~5 HZ).

The NUC is installed in ubuntu 18.04 with kernal 5.0.0-37-generic. The NUC model is 5i7RYH. Is this behavior caused by the computer?

DarthBubi, 2019-12-19:
I can confirm the same behaviour on my machine. For me a downgrade to librealsense 2.30 and recompiling the ROS driver with that version did the trick. Seems to be a bug in librealsense.

doronhi, 2019-12-29:
Can you confirm this downgrade in performance using any of the librealsense tools? `rs-benchmark`, `realsense-viewer`, `rs-callback`..? Please elaborate if you did.

armandok, 2020-01-30:
I am having the same issue with 2.32.1. My kernel version is 5.3.0-7625-generic running on Ubuntu 18.04 and I'm using pre-build packages. Does anyone know how to downgrade librealsense to 2.30 with apt-get?

doronhi, 2020-02-02:
[#5751](https://github.com/IntelRealSense/librealsense/pull/5751) seems to fix the issue for me on Ubuntu 18.04, kernel 4.15.0-74-generic

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@armandok Did you try https://github.com/IntelRealSense/librealsense/pull/5751 to see if it can fix your issue? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@armandok  Any update from your side? Thanks!

armandok, 2020-02-17:
@RealSenseCustomerSupport Yup. Tried it and fixed the problem! Thanks!

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@armandok Glad to hear that. Thanks for the feedback!

closed 2020-02-17 by RealSenseCustomerSupport