realsenseai/realsense-ros #1025: Can D435 filter the image for modeling according to specific IMU data？

TomLuosy, 2019-12-16:
Hi, I want to make d435 choose whether to transmit the corresponding image information according to the specific IMU data, so as to save the bandwidth of information transmission.

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

@TomLuosy Could you please specify more about "transmit the corresponding image information according to the specific IMU data"? What kind of specific IMU data? Thanks!

TomLuosy, 2020-01-04:
Thank you very much for your reply. I put two d435 sensors on the robot. Because I wanted to model the whole environment around the robot, I added periodic rotating devices and placed sensors on them. But the modeling effect of the sensor in the process of rotation is not very ideal, so I want to set the rotation device to stop at a specific angle, and model at the moment of stop. The advantage of this method is to save bandwidth and avoid transmitting image information of low quality during rotation.
I wonder if my idea can be realized? Or is d435 not suitable for rotation when it is used for modeling?

RealSenseCustomerSupport, 2020-01-09:
----------------------------------------------

@TomLuosy
the IMU, RGB and depth can enable/separately..
how about  streaming IMU continuously but  only streaming depth/RGB when the rotator stop

and since the D435 use global shutter camera, larger exposure time may cause blur image which the camera is moving. you can adjust the exposure time to see if it can help on your case.

TomLuosy, 2020-01-09:
Thank you very much for your advice, which has brought me a lot of help. I found that there is no exposure time setting in my realsense version, so I am updating my realsense version, hoping to add the exposure time setting to the sending command.
At the same time, I would like to ask again, which part of the code should I add statements about the opening and closing depth, RGB setting and judging IMU data? I am a novice in C + +, I hope I could get your help.
Thank you again for your advice!Happy new year！

closed 2020-02-24 by RealSenseCustomerSupport