realsenseai/realsense-ros #1026: How to find the orientation of a flat object using realsense D435 ???

abhishekpg01, 2019-12-16:
Hi i want to find the orientation of a flat plate using realsense camera. I know 4 points on the plate. Using that is there any way to find the orientation ??

MartyG-RealSense, 2019-12-16:
In computer vision, the orientation of an object can be described as its "pose".  Though pose support is not built in to the D435 like it is for the T265 Tracking Camera, RealSense users have found workarounds to get pose with a D435 using OpenCV or or inference in ROS.

A user who was using a robot arm for bin picking got pose with a D435 with an OpenCV SolvePnP algorithm.

https://github.com/IntelRealSense/librealsense/issues/3569#issuecomment-475896784

The link below, meanwhile, has an approach to getting object pose with D435 and ROS.

https://github.com/NVlabs/Deep_Object_Pose/issues/10

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

If you're asking about the orientation of the plate with the camera mounted on it then MartyG-Realsense is correct. If you mean knowing the orientation of the plate using the depth values from these 4 points that the depth camera is looking at then that is not supported by our real-sense API.

Will you be needing further assistance with this?

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-04-05 by RealSenseCustomerSupport