realsenseai/realsense-ros #1027: D435 - time stamp issue in image/depth streams

ManuelS-BSH, 2019-12-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Ubuntu 18.04.3 LTS  | 
| Kernel Version (Linux Only)    |  5.0.0-37-generic                                         | 
| Platform                                 | Notebook  |
| SDK Version                            |  2.31.0                          | 
| ROS Wrapper Version        | 2.2.11          |
| Language                            |  C++ / ROS   |
| Segment			|  Robot                          | 

### Issue description
After launching a camera stream, the time stamp of the image and depth streams is more or less frozen.
It jumps forth and back for +- 5-10 secs, but basically it's doesn't count forward as it normally should.

I've tested it with `/camera/color/image_raw`, `/camera/depth/color/points` of rs_camera.launch and with `/camera/depth_registered/points` of rs_rgbd.launch.
All of them had frozen time stamps within `rqt_topic`.

It just seems to affect the time stamp though, as RViz can show the camera stream and/or Pointclouds fluently.

doronhi, 2019-12-17:
I try to reproduce using the following commands:
Terminal 1: `roslaunch realsense2_camera rs_camera.launch`
Terminal 2: `rostopic echo /camera/color/image_raw/header/stamp`

The time stamps in terminal 2 seems fine..
I use D435, firmware version: 05.11.01.100 on Ubuntu 16.04

ManuelS-BSH, 2019-12-19:
I honestly have absolutely no idead why, because I didn't change _anything_, yet it seems to work again now, just a few days later.
Maybe the PC had issues the last time or something, I really don't know ...

doronhi, 2019-12-29:
Well, good thing is that it works now. Can we close the issue? (Please reopen if problem returns)

closed 2020-01-10 by RealSenseCustomerSupport