realsenseai/realsense-ros #1029: rectified vs non-rectified streams

JadTawil-theonly, 2019-12-16:
After running the launch file:

roslaunch realsense2_camera rs_rgbd.launch

I am confused by the fact that there is a topic /camera/color/image_raw and a topic /camera/color/image_rect_color, however, the intrinsic parameter published on /camera/color/camera_info, which i assume corresponds to the /camera/color/image_raw topic, states that the distortion parameters are 0. This means that the images published on  /camera/color/image_raw are rectified, which leads to the question, what is the difference between  /camera/color/image_raw and  /camera/color/image_rect_color ??

jokla, 2020-02-05:
I have the same question. Have you figured it out @JadTawil-theonly ?

JadTawil-theonly, 2020-02-05:
Not really, the images are the same.. rectification is happening on the device it seems

RealSenseSupport, 2020-12-05:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

doronhi, 2020-12-14:
The "/camera/color/image_rect_color" topic is the product of ros package rgbd_launch.

closed 2021-01-15 by RealSenseSupport