realsenseai/realsense-ros #1030: realsense-ros for ros melodic

ashirsat, 2019-12-17:
Hi,

I am planning to upgrade ros-kinetic to ros-melodic since it is almost end of life and ubuntu 18.04 is the next LTS.

Is there a plan for upgrading the realsense-ros driver to work with ros-melodic

Thanks.

jediofgever, 2019-12-19:
hi, @ashirsat ,
I did built the repo under melodic, 
see my issue here; https://github.com/IntelRealSense/realsense-ros/issues/1021

ashirsat, 2019-12-19:
Hey @jediofgever ,

I will try and build it with the guide for ros-kinetic and add this extra repo.

Thanks

tylerjw, 2019-12-22:
While this will build `rosdep install` fails making it impossible to depend on this and get our repo to pass CI in Melodic.

RealSenseCustomerSupport, 2020-03-11:
----------------------------------------------

@ashirsat The latest ROS wrapper version supports ROS Melodic on Ubuntu 18.04 now. Please have a try and look forward to your update. Thanks!

RealSenseCustomerSupport, 2020-03-17:
----------------------------------------------

@ashirsat Any update from your side? Thanks!

ashirsat, 2020-03-17:
I will try it out soon and update you.

Regards,
Aniket Shirsat

RealSenseCustomerSupport, 2020-03-26:
----------------------------------------------

@ashirsat Did you get chance to try? Looking forward to your update. Thanks!

ashirsat, 2020-03-27:
Hi All,

I am in the process of trying out the build on rod-melodic. Currently I am
trying to figure out another issue with px4 and ros package.

I can give an update by middle of the next week.

Sorry for the delayed response.

Regards,
Aniket

-- 
Regards,
Aniket Shirsat

Mechanical Graduate
Arizona State University| Ira A. Fulton Schools of Engineering
School for Engineering of Matter Transport and Energy
Cell:(480)371-8843
Portfolio <http://seelio.com/aniketshirsat>
<http://www.linkedin.com/in/shirsataniket>
<https://twitter.com/intent/follow?original_referer=https%3A%2F%2Fabout.twitter.com%2Fresources%2Fbuttons&region=follow_link&screen_name=AniketRShirsat&tw_p=followbutton>

RealSenseCustomerSupport, 2020-03-27:
----------------------------------------------

@ashirsat No problem. Thanks for the reply!

RealSenseCustomerSupport, 2020-04-07:
----------------------------------------------

@ashirsat Any update from your side now? Thanks!

ashirsat, 2020-04-08:
No update yet. Sorry for the delay. If required close the issue for now and
I can open it again if the issue persists.

-- 
Regards,
Aniket Shirsat

Mechanical Graduate
Arizona State University| Ira A. Fulton Schools of Engineering
School for Engineering of Matter Transport and Energy
Cell:(480)371-8843
Portfolio <http://seelio.com/aniketshirsat>
<http://www.linkedin.com/in/shirsataniket>
<https://twitter.com/intent/follow?original_referer=https%3A%2F%2Fabout.twitter.com%2Fresources%2Fbuttons&region=follow_link&screen_name=AniketRShirsat&tw_p=followbutton>

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@ashirsat Thanks for the feedback! Will close this accordingly. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-04-08 by RealSenseCustomerSupport