realsenseai/realsense-ros #1033: HW Sync Issue/ Bug ?

abhileshborode, 2019-12-20:
I am working with 2 D435is and trying to get a panoramic images of the input stream. To do this successfully I am trying to HW sync the clocks of the 2 D435is so as to make sure the images are taken at the same timestamp. I followed the instructions on the whitepaper 
(https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_Multiple_Camera_WhitePaper.pdf) 
and wired them up using the sync cable as per instruction. 
Made one camera the master and the other the salve using `inter_cam_sync_mode` dynamic parameters. 

The issue is when I try to test the HW sync between 2 cameras by recording a rosbag from the 2 camera topic I get 

` /camera2/infra1/image_rect_raw        3073 msgs @  14.1 Hz : sensor_msgs/Image`
 `/camera3/infra1/image_rect_raw        3038 msgs @  14.2 Hz : sensor_msgs/Image`          

Both the cameras are publishing at different frequencies and have different number of messages.

I have already been through issues  #4653 ,  #984 which kind of cover the same issues. However there was no concrete way to make sure the HW sync was sucessfull. 

Am I missing something to make the clock of the cameras synced (HW synced)? Or is there anything wrong in which I am testing the HW sync ? 

My work station: 
Melodic 18.04. (ROS)
librealsense - 2.31

doronhi, 2019-12-29:
It is possible for the slave camera to miss frames sometimes or to have frame drops.
You should first check if the timestamps from both cameras are roughly the same.
Then you should check whether images with very close timestamps from both cameras were taken at the same time. This is best done with LED-panel but you could also check with some moving object in view, like a rotor rotating at 1Hz or something similar.

closed 2020-01-20 by RealSenseCustomerSupport

Andyshen555, 2020-10-18:
How do you set the inter_cam_sync_mode through launch file? Is there a way to do so?

abhileshborode, 2020-10-19:
Add this to your launch file
```
  <arg name="dynamic_config" default="dynamic_parameter.yaml"/>
    <node name="dynamic_reconfigure_load" pkg="dynamic_reconfigure" type="dynparam"
          args = "$(arg dynamic_config)" />
```
set inter_cam_sync_mode: 1 in the yaml file

jesseArcbyt, 2021-07-19:
hello @abhileshborode sorry to bother you on an old issue, but I have a quick question. In your comment, you mention "set inter_cam_sync_mode: 1 in the yaml file".  

What yaml file are you referring to and where is it located?  I can make my own yaml file but ROS can never locate it. 

Thanks!

abhileshborode, 2021-07-19:
Make your own yaml file and you can load it using something like
```
    <node name="dynamic_reconfigure_load" pkg="dynamic_reconfigure" type="dynparam"
	  args="load / $(find package)path_to_yaml_file/$(arg dyn_config)" />
```