realsenseai/realsense-ros #1034: Depth values differ between ROS and API

KyleJosling, 2019-12-20:
I have been attempting to get accurate depth values from the realsense-ros `/camera/depth/aligned_depth_to_color/image_raw` topic, but I have found that the depth values I am getting are different than the ones I am getting from the API as well as `realsense-viewer`. The ones from ROS are not accurate to the actual depth.

I have set up a test program to read depth values from a rs2 depth stream, an example is shown below:
```
    rs2::pipeline pp;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16 , 30);
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_RGB8, 30);
    auto MyPipelineProfile = pp.start(cfg); 

    rs400::advanced_mode MyDevice = MyPipelineProfile.get_device();
    rs2::depth_sensor DepthSensor = MyDevice.first<rs2::depth_sensor>();
    float DepthUnits = DepthSensor.get_option(RS2_OPTION_DEPTH_UNITS);

    auto DepthStream = MyPipelineProfile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();

    float depth = depthFrame.get_distance(352, 216);
```

In ROS, I am using cv_bridge and converting the depth frame to a `cv::Mat` and getting a value from it, similar to what is shown below:

```
depthFrame = cv_bridge::toCvCopy(depthImageMsg, sensor_msgs::image_encodings::TYPE_16UC1)->image;
unsigned short depth = depthFrame.at<unsigned short>(352, 216);
```
But the values differ by around 3 cm. I have been subscribing to the `/camera/aligned_depth_to_color/image_raw`, but I have also tried the `/camera/depth/image_rect_raw` topic as well.

Can anyone help me with this? Thanks in advance

RealSenseCustomerSupport, 2020-03-12:
----------------------------------------------

@KyleJosling Sorry for late response! Could you please specify what's the SDK version and ROS wrapper version when you get the issue? And what's your ubuntu and kernel version. Did you get chance to try the latest SDK and ROS wrapper to see if you still have the same issue? Thanks!

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

@KyleJosling Could you please update? Thanks!

KyleJosling, 2020-03-23:
Hello,

Turns out OpenCV's function to get the value from a matrix is cv::Mat::at(int row, int col), where realsense's is get_distance(int x, int y). I should have been using:

```
depthFrame = cv_bridge::toCvCopy(depthImageMsg, sensor_msgs::image_encodings::TYPE_16UC1)->image;
unsigned short depth = depthFrame.at<unsigned short>(216, 352);
```

I just had the coordinates flipped. Silly mistake on my part.

Closing this issue.

closed 2020-03-23 by KyleJosling