realsenseai/realsense-ros #1035: color cam scale is smaller than depth cam scale ?

jediofgever, 2019-12-24:
Hi there, 
I was trying newly acquired D435, I find out the depth image covers a wider area than color camera although i set their resolution identical , is this expected behavior ?, how can I make them exactly to correspond pixel by pixel ? 
Thank you very much

MartyG-RealSense, 2019-12-24:
I hope that the linknk below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/2279

jediofgever, 2019-12-24:
@MartyG-RealSense , Hi , thanks for your reply 

I saw the direction that link suggests, however aligning depth and color programmatically in the same depth camera is not efficient and logical, also I assume after alignment I will just get a cropped depth image since FOV of color camera cant be changed, am I right ? 
if this is the case then how RGB-D is acquired, would not that cause RGB-D point clouds to have less precise RGB values ?

MartyG-RealSense, 2019-12-24:
Aligning depth and color in the same camera is a standard approach for RealSense.  If you find that the process is too slow for your liking then you can speed it up by offloading processing from the CPU to the GPU.

This can be done by building Librealsense with CUDA support if you have an Nvidia GPU, or with GLSL processing blocks if you do not have an Nvidia GPU.  The details for doing so can be found in the link below.

https://github.com/IntelRealSense/librealsense/issues/5440#issuecomment-565124976

Your RGB-D question is outside of my knowledge unfortunately, so I will refer it to one of the Intel staff on this forum.  @doronhi

doronhi, 2019-12-29:
Hi @jediofgever ,
The Color camera has narrower FOV than both infra cameras that create the depth image. Therefor an aligned depth to color image will indeed be cropped.
What is the RGB-D image and pointcloud that you refer to? Specifically, what topics and launch file are you referring to?

jediofgever, 2020-01-06:
hi, @doronhi  
by RGB-D I refereed to colorful point cloud, I am not sure if the extreme poor performance of point cloud is caused by this alignment. It can hardly deliver 1 frame of point cloud in 5-10 seconds. I built librealsense with CUDA support, I have nVIDIA 1080 TI gpu as well as bunch of  RAM.

closed 2020-01-06 by jediofgever