realsenseai/realsense-ros #1036: nvidia jetson nano occur error

sss22213, 2019-12-26:
hi there:
  i use realsense-ros  occur error or warning.

OS: Ubuntu 18.04
L4T: 32.3.1
RealSense Firmware: 5.12.0
realsense-ros: 2.2.11
LibRealSense v2.31.0
Device: d435i

[ WARN] [1577332384.972041372]: Error extracting usb port from device with physical ID: 2-1.2-7
Please report on github issue at https://github.com/IntelRealSense/realsense-ros

WARNING [546844950912] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

tmichals, 2019-12-31:
I am having the same problem https://github.com/IntelRealSense/realsense-ros/issues/1039... and if you look we are the only issues that have not been answered.

swinterbotix, 2020-01-09:
Running into the same issue - also using the Jetson Nano

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

Hi @sss22213,

The issue was addressed in this PR:
https://github.com/IntelRealSense/realsense-ros/pull/1088
Please let us know, if you need any further assistance with this matter.

Thank you!

aswinkumar2019, 2020-02-23:
Guys,
Were you able to generate slam using jetson nano?

tmichals, 2020-02-24:
Yes,  we were able to publish our T256 pose information using the realsense rs_T265 launch file, but in comparing why the T265 and d435 create usb errors appears to come down to the "publish_odom_tf" statement.   Remove that statement and both camera's will publish their pose and pointcloud data.

aswinkumar2019, 2020-02-24:
Hello tmichals,
Was the map generated successfully?,how did you install ros realsense?, through jetson hacks script or, from source or, through apt?
Regards,
Aswin Kumar

tmichals, 2020-02-24:
With some effort we have transforms that created at single tree using the odom T265 as a base and tf to the d435, but we have not gotten rtabmap to work with are published data.   Our understanding is to use the t265 for location while combining that with d435 and its distance point cloud.   We were able to write a rospy application that could access the published data.    Yes, we used jetsonhacks scripts to install the everything, including the realsense wrapper.

aswinkumar2019, 2020-02-24:
Why don't you use MPU 9250 IMU in place of t265 for localisation?
Hope d435 and MPU 9250 is enough to generate slam. 
Regards, 
Aswin Kumar

sss22213, 2020-02-27:
hi RealSenseCustomerSupport:
Problem still exist....

[image]

[image]

aswinkumar2019, 2020-02-27:
Hello sss22213,
Did you use the source of the latest repo?
I didn't check it as I was busy on jetson nano and lidar. 
Can you please verify it and reply? 
Regards, 
Aswin Kumar

sss22213, 2020-02-27:
Hello Aswin Kumar,
i use 9c447a2f90e8c479111412eedf8e830090ad5b78

[image]

[image]

Regards,
sss22213

aswinkumar2019, 2020-02-27:
Try rebooting the system, if not fixed, try to run as root user.(you have to source the setup.bash file again). 
The control transfer error was gone for me.
Is the error extracting usb port gone? 
Regards, 
Aswin Kumar

tmichals, 2020-02-27:
Would this solution be similar to [#1037](https://github.com/IntelRealSense/realsense-ros/issues/1037#event-3075588915)

aswinkumar2019, 2020-02-27:
Yeah, it is the same issue.
Guys, you need to reduce the rate parmater which would be 300 in launch file to a lower value(in robot_localization) node.
Otherwise, you will get a lot of warnings in jetson nano. 
I got the point cloud but, didn't get the map because of these errors. 
I just reboot and run the launch file as root user while keeping the realsense in a optimum position and map was generated but, after some time, I got the same error and it stopped. 
If anybody find the solution for this please reply

RealSenseCustomerSupport, 2020-02-29:
----------------------------------------------

@sss22213,

In the description of the issue you didn't mention that your system is running on a virtual machine. We don't support installations on VM. Can you please use non-virtual environment and let us know whether or not you experience the same issue on it as well?

aydal, 2020-03-02:
Hello. 
I am having the same issue working on jetson nano. Point clouds are not being formed. Tried running the code as root too

arjunsengupta1998, 2020-03-02:
@aydal  did you find a solution? I am having the same issue. 
Using D435 on Jetson Nano. I am able to view all color and depth images on rviz but am unable to view depth cloud.

To elaborate, 

I first run the command:

`roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`
 
After setting up initial processes, the terminal returns the following warning:

>  [ WARN] [1582923381.445252274]: Error extracting usb port from device with physical ID: 2-1.4-5
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
 
Following that, initial processes run and the realsense node is set-up. Immediately after it starts running, the following warning is displayed:

> 28/02 12:56:28,596 WARNING [546937237888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/02 12:56:28,657 WARNING [546937237888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 28/02 12:56:28,722 WARNING [546937237888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

As I run rviz and add the image topics, I am able to view all RGB, depth and IR images. 
But as I try to add the pointcloud2 topic, the terminal throws the following messages:

> [ERROR] [1582924070.443140553]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
 28/02 13:07:50,705 WARNING [546559730048] (types.cpp:49) Out of frame resources!
 28/02 13:07:50,705 ERROR [546559730048] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!

And no pointcloud is visible on rviz either.

As suggested on one of the threads, I tried setting the fps parameters to 15 in the launch file as well. It still doesn't work.

@RealSenseCustomerSupport your help would be appreciated. 

@aswinkumar2019  and @tmichals  it would be great if any of you could help out as well.

aswinkumar2019, 2020-03-02:
Guys,
I atleast got point cloud along with these errors in old version without this pull request. 
I tried using the latest branch, I am not getting even the point cloud. 
Please fix it guys. 
For others, guys please try older branches, I was able to get point cloud. 
I think something is messed up in latest versions. 
Regards, 
Aswin Kumar

aswinkumar2019, 2020-03-02:
I didnt get these errors in older versions.
All these errors are only in newer branches

arjunsengupta1998, 2020-03-04:
Now I am getting the point cloud, but the errors still persist. I simply set the initial_reset parameter to true on launch. However, these errors are still there and the frame rate is extremely irregular.

doronhi, 2020-03-05:
#1097 by @facontidavide was just merged. Could you check if it helps with the irregular frame rate as it may be caused by lack of resources.

jorgemia, 2020-03-07:
@arjunsengupta1998 @doronhi  I was having the same warnings (error extracting usb... and control_transfer error) I managed to fix these by applying PR #1088 (that didn't fix it immediately) and then running `sudo apt purge ros-kinetic-librealsense2` as suggested in #1051 . After this, I ran catkin_make and no longer get issues when setting up the ros node.

I can see the pointcloud in rviz and even though I applied the improvements in #1097 I occasionally get the following errors when visualising PointCloud2 and image (from /image_raw) in rviz.
```
07/03 16:42:07,414 WARNING [140361804539648] (types.cpp:49) Out of frame resources!
07/03 16:42:07,414 ERROR [140361804539648] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!
```

Is this due to irregular frame rate?

RealSenseCustomerSupport, 2020-03-16:
----------------------------------------------

Hi @jorgemia,

Would you please elaborate on what does it mean that PR https://github.com/IntelRealSense/realsense-ros/pull/1088 doesn't fix the "Error extracting usb port" issue immediately?

Thank you!

jorgemia, 2020-03-17:
@RealSenseCustomerSupport when I applied the changes in #1088 my device continued to give the error “error extracting USB port”

RealSenseCustomerSupport, 2020-04-17:
----------------------------------------------

@jorgemia,

Based on your comment you "no longer get issues when setting up the ros node".Do you still see the "Error extracting usb port" message or it's completely gone?

jorgemia, 2020-04-17:
@RealSenseCustomerSupport no I no longer see that error on an **Up Squared** board

aswinkumar2019, 2020-04-17:
This error actually exists in jetson nano, not on upsquared.
It is not yet fixed

tmichals, 2020-04-18:
Yes I am at this moment still getting the error... https://github.com/IntelRealSense/realsense-ros/issues/1039

RealSenseCustomerSupport, 2020-04-21:
----------------------------------------------

@aswinkumar2019 and @tmichals,

Can you please confirm that you have applied this PR :
https://github.com/IntelRealSense/realsense-ros/pull/1088
and it didn't fix the issue?

Thank you!

aswinkumar2019, 2020-04-23:
yes,the issue is still there in my nano even after applying it.
Regards,
Aswin Kumar

RealSenseSupport, 2020-06-17:
@aswinkumar2019 ,

Please share all the relevant details of your system configuration (versions of L4T,ROS Wrapper, etc.)
and attach the logs and the screenshots of the error messages you see.

Thank you!

RealSenseSupport, 2020-06-23:
@aswinkumar2019 ,

Do you still need help with this matter?

aswinkumar2019, 2020-06-24:
@RealSenseCustomerSupport i applied the pull request and,got the error even after that.
Sorry,Currently i have left my job so,i dont have access to my nano or realsense anymore.
Please verify with @tmichals 
Regards,
Aswin Kumar

tmichals, 2020-06-25:
I will need time to double check, but I think that I remember being able to lower the frame size on the T265 and D435, or lower the fps on my nano ... so that I was able to over come the USB port issue.  Seems like this was about 2 months ago that I did a complete re-flash of my Nano with realsense.   Our robotics club was interupted by Covid... we should be regrouping soon.

RealSenseSupport, 2020-06-25:
@aswinkumar2019

Thank you for the update!

@tmichals

Let's close this thread for now, as we cannot move forward with the issue investigation, and re-engage it whenever you will have all the needed details.
Please let us know if you have any additional questions regarding this matter at the present moment.

closed 2020-07-02 by RealSenseSupport