realsenseai/realsense-ros #1037: Cant launch RS-415 on Jetson Nano with ROS melodic and Ubuntu 18.04

matthewp14, 2019-12-26:
I followed the tutorials below to install everything: 

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md
https://github.com/IntelRealSense/realsense-ros

I also checked all of my dependencies on the .yml file. No issues there.
https://github.com/IntelRealSense/realsense-ros/blob/development/.travis.yml

I can launch the camera using realsense-viewer and have no issues. Both the rgb and stereo modules work fine. The issues start when I try the following: 

`roslaunch realsense2_camera rs_camera.launch ` 

I get the following output (after a series of info messages including RealSense Node Is Up!):

 26/12 10:39:55,666 WARNING [547214987648] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 26/12 10:39:55,756 WARNING [547214987648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 26/12 10:39:55,819 WARNING [547214987648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 26/12 10:39:55,911 WARNING [547214987648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 26/12 10:39:56,494 WARNING [547214987648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 26/12 10:39:56,682 WARNING [547214987648] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

Additionally, if I try `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`

I get the following output: 

26/12 10:53:33,957 WARNING [547352981888] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 26/12 10:53:34,047 WARNING [547352981888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 26/12 10:53:34,142 WARNING [547352981888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 26/12 10:53:34,157 WARNING [546548863360] (types.cpp:49) Out of frame resources!
 26/12 10:53:34,157 ERROR [546548863360] (synthetic-stream.cpp:47) Exception was thrown during user processing callback!
 26/12 10:53:34,178 WARNING [546548863360] (types.cpp:49) Out of frame resources!
 26/12 10:53:34,178 ERROR [546548863360] (synthetic-stream.cpp:47) Exception was thrown during user processing callback!
[ERROR] [1577375614.196464158]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
 26/12 10:53:34,205 WARNING [547352981888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 26/12 10:53:34,210 WARNING [546548863360] (types.cpp:49) Out of frame resources!
 26/12 10:53:34,210 ERROR [546548863360] (synthetic-stream.cpp:47) Exception was thrown during user processing callback!
 26/12 10:53:34,280 WARNING [547352981888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 26/12 10:53:34,290 WARNING [546548863360] (types.cpp:49) Out of frame resources!
 26/12 10:53:34,291 ERROR [546548863360] (synthetic-stream.cpp:47) Exception was thrown during user processing callback!
 26/12 10:53:34,311 WARNING [546548863360] (types.cpp:49) Out of frame resources!
 26/12 10:53:34,311 ERROR [546548863360] (synthetic-stream.cpp:47) Exception was thrown during user processing callback!
 26/12 10:53:34,345 WARNING [546548863360] (types.cpp:49) Out of frame resources!
 26/12 10:53:34,346 ERROR [546548863360] (synthetic-stream.cpp:47) Exception was thrown during user processing callback!
 26/12 10:53:34,377 WARNING [546548863360] (types.cpp:49) Out of frame resources!
 26/12 10:53:34,377 ERROR [546548863360] (synthetic-stream.cpp:47) Exception was thrown during user processing callback!
 26/12 10:53:34,377 WARNING [547352981888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 26/12 10:53:34,409 WARNING [546548863360] (types.cpp:49) Out of frame resources!
 26/12 10:53:34,410 ERROR [546548863360] (synthetic-stream.cpp:47) Exception was thrown during user processing callback!

If you could please provide some guidance on this that would be great! Thank you

yyxr75, 2019-12-27:
Here I came across the same issue as yours.
Do you have any idea about solution of it?
The topics were published as they can be shown after
`rostopic list`, but show no imges when using `rviz` to subscribe.
All I have is warnings and errors.
any hint would be helpful if you would like to share!
so much appreciation!

btw, the [website](https://www.jetsonhacks.com/2019/10/25/realsense-ros-wrapper-jetson-nano/) was talking useless shit

tmichals, 2019-12-31:
This looks like what I am also seeing.. https://github.com/IntelRealSense/realsense-ros/issues/1039.  Do you agree?

richardrl, 2020-01-04:
We're seeing the same error. 

04/01 20:57:13,713 WARNING [139805191870208] (types.cpp:49) Out of frame resources!
 04/01 20:57:13,713 ERROR [139805191870208] (synthetic-stream.cpp:47) Exception was thrown 
during user processing callback!

KevinHanwang, 2020-01-13:
> We're seeing the same error.
> 
> 04/01 20:57:13,713 WARNING [139805191870208] (types.cpp:49) Out of frame resources!
> 04/01 20:57:13,713 ERROR [139805191870208] (synthetic-stream.cpp:47) Exception was thrown
> during user processing callback!

Me too, but I don't know how to solve this problem

KevinHanwang, 2020-01-14:
I solved this problem right now. Open your launch file, and modify the frame rate parameter. Your default value is 30, you can make it smaller.
```
  <arg name="depth_fps"           default="15"/>
  <arg name="infra_fps"           default="15"/>
  <arg name="color_fps"           default="15"/>
```

matthewp14, 2020-01-14:
@KevinHanwang I tried that but am still getting the same issue. I should note that the first warning is:

Error extracting usb port from device with physical ID: 2-1.1-4

KevinHanwang, 2020-01-15:
> @KevinHanwang I tried that but am still getting the same issue. I should note that the first warning is:
> 
> Error extracting usb port from device with physical ID: 2-1.1-4

Maybe it's the computer hardware. You can try another computer

LEGO999, 2020-01-16:
Could be the same issue here [https://github.com/IntelRealSense/librealsense/issues/5659](url)
, 18.04 LTS, Melodic, only difference is, i am using RPi4.

RealSenseCustomerSupport, 2020-02-20:
----------------------------------------------

Hi matthewp14,

Just to make sure : did you build both LibRealSense and ROS Wrapper from the source code?

Thank you!

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@matthewp14,

Do you still need help with this question?

matthewp14, 2020-02-26:
@RealSenseCustomerSupport I was able to get around this for what I was doing. The errors are still there but I no longer need a solution. And yes I built both from source

closed 2020-02-26 by matthewp14

aydal, 2020-03-02:
@matthewp14 How did you solve this?

arjunsengupta1998, 2020-03-02:
@RealSenseCustomerSupport I am facing this issue and would appreciate some help. 
Using D435 on Jetson Nano. I am able to view all color and depth images on rviz but am unable to view depth cloud.

To elaborate,

I first run the command:

roslaunch realsense2_camera rs_camera.launch filters:=pointcloud

After setting up initial processes, the terminal returns the following warning:

[ WARN] [1582923381.445252274]: Error extracting usb port from device with physical ID: 2-1.4-5
Please report on github issue at https://github.com/IntelRealSense/realsense-ros

Following that, initial processes run and the realsense node is set-up. Immediately after it starts running, the following warning is displayed:

28/02 12:56:28,596 WARNING [546937237888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
28/02 12:56:28,657 WARNING [546937237888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
28/02 12:56:28,722 WARNING [546937237888] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61

As I run rviz and add the image topics, I am able to view all RGB, depth and IR images.
But as I try to add the pointcloud2 topic, the terminal throws the following messages:

[ERROR] [1582924070.443140553]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
28/02 13:07:50,705 WARNING [546559730048] (types.cpp:49) Out of frame resources!
28/02 13:07:50,705 ERROR [546559730048] (synthetic-stream.cpp:48) Exception was thrown during user processing callback!

And no pointcloud is visible on rviz either.

As suggested above, I tried setting the fps parameters to 15 in the launch file as well. It still doesn't work.

@matthewp14 I'd appreciate if you could provide some inputs as well.

matthewp14, 2020-03-04:
@aydal @arjunsengupta1998 I was never able to solve the warnings that I posted above. I too changed the parameters to reduce the frame rate but the errors remained persistent. For what I was working on I did not need to use the point cloud so I just continued on without it. Sorry that I cannot be any help to you with this.

arjunsengupta1998, 2020-03-04:
@matthewp14 thanks for the response. 
I actually was able to get the pointcloud. I simply set the initial_reset parameter to true, and got the result.
However, the warnings still persist and there are occasional irregular lags in live streaming. Also, despite the frame rate being set to 15fps, the visualized output is much slower (probably around 10 fps).

@RealSenseCustomerSupport is there any way to attain robustness?

jorgemia, 2020-03-06:
@arjunsengupta1998 I have similar warnings when trying to run` roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`on an Up Squared board with Ubuntu 16.04, Librealsense 2.33.1, ROS Kinetic and the D435i. Realsense-viewer works fine. Any ideas on how to fix it yet?