realsenseai/realsense-ros #1039: T265 and D435 launch issue. Error extracting usb port from device with physical ID: 2-1.3-6

tmichals, 2019-12-26:
Nvidia Nano JetPack 4.3 - L4T 32.3.1
LibRealSense v2.31.0
RealSense ROS v2.2.11
T265 and d435 Intel Camera

Issue:
Multiple errors:
[7692] [E] TrackingManager: Error while loading image - libusb_claim_interface failed (0xFFFFFFFA)

ERROR [547566363008] (uvc-streamer.cpp:136) failed to submit UVC request, error: -13

WARNING [547331498368] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16
____________________________________________________
I am using a launch file based on the rs_d400_and_t265.launch file attached 7220.txt .   I am getting a RealSense error that I think was supposed to be fixed in 2.31.    Can you tell me what I need to do to fix these issues.    My goal is simply creating ROS nods for the d435 distance and t265 tracking data for rtabmap. 

[T265 and D435 launch issue. Error extracting usb port from device with physical ID- 2-1_3-6.txt](https://github.com/IntelRealSense/realsense-ros/files/4003110/T265.and.D435.launch.issue.Error.extracting.usb.port.from.device.with.physical.ID-.2-1_3-6.txt)

[7220.launch.txt](https://github.com/IntelRealSense/realsense-ros/files/4003125/7220.launch.txt)

RealSenseSupport, 2020-09-17:
Thank you for highlighting the multi-cam launch issue with T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue for the T265.

tmichals, 2020-09-18:
Not sure why you are saying this since Intel is still selling the T265 and I do not see them selling an upgraded tracking camera?

closed 2021-03-31 by RealSenseSupport