realsenseai/realsense-ros #1040: How to increase pointclould frame rate

wikpunGitHub, 2019-12-30:
Hi, is there a way to increase the pointcloud frame rate? I set my D435 camera to have 60 fps color and 90 fps depth image. But the pointcloud is still 30 fps. Is there a way to increase the pointcloud fps? Thanks!

doronhi, 2019-12-30:
Do you get the depth and color frames at the rate you expected? (`rostopic hz /camera/color/image_raw`, `rostopic hz /camera/depth/image_rect_raw`)

If you do then the problem may be with computer power. What kind of machine do you use?

If you don't, do the followings:
1. Set the "infra_fps" to the same frame rate as the depth frame rate or disable both infra1 and infra2 completely (`enable_infra1:=false enable_infra2:=false`).
2. Turn off the auto exposure priority. You can do that by adding the followings to your launch file: 
```
  <rosparam>
      /camera/rgb_camera/enable_auto_exposure: true
      /camera/rgb_camera/auto_exposure_priority: false
  </rosparam>
```

P.S. 
There is no point in setting depth frame rate to be higher then the color frame rate as the point cloud is built using a set of the 2.

wikpunGitHub, 2019-12-30:
Hi doronhi, you are right, it is a computer power issue. I am running it on a Jetson nano. Thanks for the help!

closed 2019-12-30 by wikpunGitHub