realsenseai/realsense-ros #1041: Could I get both gyroscope and accelemeter topic more than 100hz?

jinwandou, 2019-12-30:


MartyG-RealSense, 2019-12-30:
The FPS rates of the T265 are fixed.

https://github.com/IntelRealSense/librealsense/issues/4103#issuecomment-497221237

https://github.com/IntelRealSense/librealsense/issues/4592#issuecomment-519062226

There are workarounds that some users have successfully applied to achieve custom FPS rates though.  This includes:

-  Getting every nth frame instead of all frames in order to simulate a certain frame rate.
- Using some form of timer to achieve the desired frequency.

In September 2019,  one user's workaround that worked for them was to develop a RealSense ROS node that is controlled by a 'rate'   The link below provides an explanation about rates in ROS.

https://roboticsbackend.com/ros-rate-roscpy-roscpp/ 

Another user achieved the same effect by setting up a timer in QT instead of using ROS.  The timer fired every 100 ms to get a 10hz frequency.

RealSenseCustomerSupport, 2020-01-23:
----------------------------------------------

Thanks MartyG-RealSense for the links and info

Hi @jinwandou
Is there anything further needed on this thread?  The FPS rates for T265 are indeed fixed.

closed 2020-02-05 by RealSenseCustomerSupport