realsenseai/realsense-ros #1042: Could I publish some topics in part when I use d435 and t265 ?

wol01179, 2019-12-30:
Hi, I am currently using RealSense d435 and t265. 
When I utilize rolaunch realsense2_camera rs_rgbd.launch and rolaunch realsense2_camera rs_t265.launch, a number of topics will appear.  
However, I need only two topics,  one topic for pointcloud2 through d435 and one topic for odometry through t265. 
Could you let me know if there is a way to effectively reduce the topic?
Thank you for reading.

doronhi, 2019-12-30:
You can reduce some of the topics by disabling sensors. You could set to false the following variables in the D435's launch file: "enable_infra1", "enable_infra2".
The rest of the topics are published but do nothing until you actually subscribe to them.

wol01179, 2019-12-31:
Thank you for your answers and good solutions. Through the method you suggested, I was able to effectively reduce the number of topics.
However, I have additional questions.
If so, Is the depth topic(reduced by the proposed method) the data that does not include the data of  infra projectors?

Additionally, I could set to false the following variables in the T265's launch: "able_fisheye1", "enable_fisheye2" 
In this case, Is the published topic(related to odometry) the data that does not use values of fisheyes ?
If so be, Is the published topic(related to odometry) only based IMU data ? 
   
Thank you for reading it.

RealSenseCustomerSupport, 2020-01-31:
----------------------------------------------

Hi @wol01179
Were you able to implement the changes outlined by doronhi?  Depth topic should include depth data.  With regards to T265 the readme outlines the topics available https://github.com/IntelRealSense/realsense-ros, the odometry will provide pose data even if you don't want the fisheyes.  The SLAM algo uses the data via the fisheyes and IMU to produce the pose data.  You can subscribe to IMU specific data.

I think we can close this thread out.  Let us know if anything further is needed here.

closed 2020-03-23 by RealSenseCustomerSupport