realsenseai/realsense-ros #1044: issue while using opensouce_tracking.launch on raspbery pi 3b with a D435

LittleMoonMan, 2019-12-31:
I am trying to make a map for my robot with use of the D435.
the guide that i am folowing tells me i need to use opensource_tracking.launch
when i use this launch file rivz start up like normal but i dont get a immage (or anything else for that matter) the only moment i get a pointcloud is when i use the demo_pointcloud.launch file. i am new with ROS and realsense so it could be a real dum mistake.

here some more information that might clarify the situation
`ruudstaal@ruudstaal-desktop:~/emely_ws$ roslaunch realsense2_camera opensource_tracking.launch
... logging to /home/ruudstaal/.ros/log/4ddeccd2-2bd1-11ea-9431-b827ebcfb969/roslaunch-ruudstaal-desktop-10378.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ruudstaal-desktop:36198/

SUMMARY
========

CLEAR PARAMETERS
 * /ukf_se/

PARAMETERS
 * /ImuFilter/_publish_tf: False
 * /ImuFilter/_world_frame: enu
 * /ImuFilter/use_mag: False
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
[... 257 lines cut ...]
[ WARN] [1577802771.182278260]: Hardware Notification:USB SCP overflow,1.5778e+12,Error,Hardware Error
[ INFO] [1577802771.765002133]: RGBDOdometry: approx_sync    = true
[ INFO] [1577802771.765674891]: RGBDOdometry: queue_size     = 10
[ INFO] [1577802771.766220931]: RGBDOdometry: subscribe_rgbd = false
[ INFO] [1577802771.766667700]: RGBDOdometry: rgbd_cameras   = 1
[ INFO] [1577802772.083359704]: 
/rtabmap/rgbd_odometry subscribed to (approx sync):
   /camera/color/image_raw,
   /camera/aligned_depth_to_color/image_raw,
   /camera/color/camera_info
[ WARN] [1577802773.186436892]: Hardware Notification:USB SCP overflow,1.5778e+12,Error,Hardware Error
[rtabmap/rgbd_odometry-4] process has died [pid 10401, exit code -11, cmd /home/ruudstaal/emely_ws/devel/lib/rtabmap_ros/rgbd_odometry --delete_db_on_start rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image_relay odom:=odom __name:=rgbd_odometry __log:=/home/ruudstaal/.ros/log/4ddeccd2-2bd1-11ea-9431-b827ebcfb969/rtabmap-rgbd_odometry-4.log].
log file: /home/ruudstaal/.ros/log/4ddeccd2-2bd1-11ea-9431-b827ebcfb969/rtabmap-rgbd_odometry-4*.log
[ WARN] [1577802775.186683771]: Hardware Notification:USB SCP overflow,1.5778e+12,Error,Hardware Error
[ WARN] [1577802777.197943948]: Hardware Notification:USB SCP overflow,1.5778e+12,Error,Hardware Error
[ INFO] [1577802778.162226268]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1577802778.167206563]: rtabmap: Deleted database "/home/ruudstaal/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[ INFO] [1577802778.167559843]: rtabmap: Using database from "/home/ruudstaal/.ros/rtabmap.db" (0 MB).
[rtabmap/rtabmap-5] process has died [pid 10402, exit code -11, cmd /home/ruudstaal/emely_ws/devel/lib/rtabmap_ros/rtabmap --delete_db_on_start rgb/image:=/camera/color/image_raw depth/image:=/camera/aligned_depth_to_color/image_raw rgb/camera_info:=/camera/color/camera_info rgbd_image:=rgbd_image_relay left/image_rect:=/stereo_camera/left/image_rect_color right/image_rect:=/stereo_camera/right/image_rect left/camera_info:=/stereo_camera/left/camera_info right/camera_info:=/stereo_camera/right/camera_info scan:=/scan scan_cloud:=/scan_cloud user_data:=/user_data user_data_async:=/user_data_async gps/fix:=/gps/fix tag_detections:=/tag_detections odom:=odom imu:=/imu/data __name:=rtabmap __log:=/home/ruudstaal/.ros/log/4ddeccd2-2bd1-11ea-9431-b827ebcfb969/rtabmap-rtabmap-5.log].
log file: /home/ruudstaal/.ros/log/4ddeccd2-2bd1-11ea-9431-b827ebcfb969/rtabmap-rtabmap-5*.log
[ WARN] [1577802779.208240123]: Hardware Notification:USB SCP overflow,1.5778e+12,Error,Hardware Error
`

[image]
[image]

MartyG-RealSense, 2019-12-31:
Are you using the guide for the D435i camera model in the link below please?

https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i

The D435 does not contain an 'IMU' tracking hardware component that the D435i ( 'i' for IMU) has.

For SLAM mapping with an IMU-less D435, the SLAM software 'ORB_SLAM2' may be a good choice.  The link below has a guide to using it with ROS.

http://wiki.ros.org/orb_slam2_ros

And here is a discussion on using it with a D435, ROS and Rviz.

https://github.com/ethz-asl/voxblox/issues/184

LittleMoonMan, 2020-01-05:
thank you for your help

i was not using a D435i but a D435. with the new package i was able to get some dots on my screen (after that my raspberry overheated). do you know sombody that was able to make a map with a raspberry or am i pushing it to a limit with it.

MartyG-RealSense, 2020-01-05:
You could try offloading some of the CPU's work onto the GPU using a Librealsense technique called GLSL Processing Blocks and see if distributing the workload reduces your Pi's temperature.

https://github.com/IntelRealSense/librealsense/pull/3654

The Pi 3 and models before it use a  VideoCore IV GPU, whilst the new Pi 4 uses a more powerful VideoCore VI GPU.

A low-power development board such as Pi may struggle with computing SLAM, even with just one camera attached (though as you found, it should be able to handle a one-camera setup).  The discussion in the link below addresses the subject of using SLAM on low-power boards.

https://github.com/IntelRealSense/librealsense/issues/3218

If your robot can accommodate a larger board and you have the budget for it, the Up Squared is an excellent powerful choice of development board to pair with a 400 Series camera.

LittleMoonMan, 2020-01-27:
in my research with the raspberry pi 4 I coudent find a version of ubuntu (mate) that was compatible with ROS. Did i mis some thing or are people using different programs?

MartyG-RealSense, 2020-01-27:
The RealSense ROS wrapper officially has compatibility with ROS Indigo (14.04) and Kinetic (16.04).  Some users have said that the wrapper's instructions work too with Melodic (18.04).  

I rarely hear of Ubuntu MATE being used (it was recommended by Intel for use with Raspberry Pi 3 Model B).  As Pi 4 is more powerful and can provide desktop-level performance, it may be beter to use one of the mainstream Ubuntu versions such as 18.04 with the ROS wrapper if possible.

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@LittleMoonMan Any other questions for this ticket? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@LittleMoonMan Any update from your side? Will close the issue if no other questions. Thanks!

closed 2020-02-25 by RealSenseCustomerSupport