realsenseai/realsense-ros #1046: Need depth, RGB and IR intensity information.

sjvinay79, 2020-01-03:
I am using the D415 camera to obtain 3D information. I need a point cloud that has RGB and intensity information from the Infrared camera. I am using a ROS workspace. I subscribe to the /camera/depth/color/points topic and I am successfully able to convert the sensor_msgs/PointCloud2 to pcl::PointCloud<pcl::PointXYZRGB> format. I am just wondering if there is any way to get IR intensity as well as RGB data in this point cloud ?

When I use pointcloud_texture_stream ="RS2_STREAM_INFRARED" as a launch parameter, I am able to get a IR textured point cloud. But for the application we are working on, we need both RGB textured and IR textured point cloud. We need a XYZ depth value that also has IR intensity information as well as a RGB values. Can we do this with the D415 ?

When I run a Ros node with the following code I get a warning saying "rgb" field is absent.

pcl::PointCloud<pcl::PointXYZRGB>::Ptr intensity_cloud(new pcl::PointCloud<pcl::PointXYZRGB>);
 pcl::fromROSMsg(*input, *intensity_cloud);

Does anybody know how to subscribe to a sensor_msgs/PointCloud2 callback and get depth, IR and RGB data ?

doronhi, 2020-01-05:
realsense2_camera is a ROS wrapper for the librealsense library.
The librealsense's set of points contains coordiante values for the aligned to depth images and these are used, in combination with the intensity images, to obtain the intensity values of the wrapper's pointcloud. You can see the process [in base_realsense_node.cpp](https://github.com/IntelRealSense/realsense-ros/blob/008e2e434a1a6523f7c9b733689273e9fa1579a7/realsense2_camera/src/base_realsense_node.cpp#L1986)
The pointcloud realsense2_camera node transmits contains either RGB or IR texture information but not both.
I'm not sure if the pointcloud structure for ROS supports having 2 sets of intensity values for every point. If so you can modify the code in case_realsense_node.cpp to your own needs. 
Otherwise you can write a code subscribing the depth, IR and color images and mimic the projection logic.

P.S.
I used the word intensity to describe both IR texture and color texture.

RealSenseCustomerSupport, 2020-01-21:
----------------------------------------------

Thanks doronhi!

Does this answer your question sjvinay?

closed 2020-01-28 by RealSenseCustomerSupport

FrankLin9981, 2021-03-09:
Hello @sjvinay79 ,

Sorry for bothering!
Could you provide the details about how you generate point cloud data with intensity field?

I follow your instruction to use **pointcloud_texture_stream ="RS2_STREAM_INFRARED"** as a launch parameter.
And the point cloud **does have the intensity field**, but the **intensity values of these points are all zeros** (10e-40).
The following are the launch parameters I use, and the launch file I use is `rs_camera.launch`.
Do I misunderstand anything?

Hope you guys could do me a favor!
Thanks a lot!

```
  <arg name="serial_no"           default=""/>
  <arg name="usb_port_id"         default=""/>
  <arg name="device_type"         default=""/>
  <arg name="json_file_path"      default=""/>
  <arg name="camera"              default="d415"/>
  <arg name="tf_prefix"           default="$(arg camera)"/>
  <arg name="external_manager"    default="false"/>
  <arg name="manager"             default="realsense2_camera_manager"/>
  <arg name="output"              default="screen"/>

  <arg name="fisheye_width"       default="-1"/>
  <arg name="fisheye_height"      default="-1"/>
  <arg name="enable_fisheye"      default="false"/>

  <arg name="depth_width"         default="640"/>
  <arg name="depth_height"        default="480"/>
  <arg name="enable_depth"        default="true"/>

  <arg name="confidence_width"    default="-1"/>
  <arg name="confidence_height"   default="-1"/>
  <arg name="enable_confidence"   default="false"/>
  <arg name="confidence_fps"      default="-1"/>

  <arg name="infra_width"         default="640"/>
  <arg name="infra_height"        default="480"/>
  <arg name="enable_infra"        default="false"/>
  <arg name="enable_infra1"       default="true"/>
  <arg name="enable_infra2"       default="true"/>
  <arg name="infra_rgb"           default="false"/>

  <arg name="color_width"         default="640"/>
  <arg name="color_height"        default="480"/>
  <arg name="enable_color"        default="true"/>

  <arg name="fisheye_fps"         default="30"/>
  <arg name="depth_fps"           default="30"/>
  <arg name="infra_fps"           default="30"/>
  <arg name="color_fps"           default="30"/>
  <arg name="gyro_fps"            default="400"/>
  <arg name="accel_fps"           default="250"/>
  <arg name="enable_gyro"         default="false"/>
  <arg name="enable_accel"        default="false"/>
  
  <arg name="enable_pointcloud"         default="true"/>
  <arg name="pointcloud_texture_stream" default="RS2_STREAM_INFRARED"/>
  <arg name="pointcloud_texture_index"  default="0"/>
  <arg name="allow_no_texture_points"   default="false"/>
  <arg name="ordered_pc"                default="true"/>

  <arg name="enable_sync"               default="true"/>
  <arg name="align_depth"               default="true"/>

  <arg name="publish_tf"                default="false"/>
  <arg name="tf_publish_rate"           default="0"/>

  <arg name="filters"                   default=""/>
  <arg name="clip_distance"             default="2.0"/>
  <arg name="linear_accel_cov"          default="0.01"/>
  <arg name="initial_reset"             default="false"/>
  <arg name="unite_imu_method"          default=""/>
  <arg name="topic_odom_in"             default="odom_in"/>
  <arg name="calib_odom_file"           default=""/>
  <arg name="publish_odom_tf"           default="true"/>
```

gcaccia1, 2022-07-16:
@MartyG-RealSense thank you for your reply! 
pointcloud_texture_stream is already set as RS2_STREAM_INFRARED.
I can see the IR intensity values attached to the pointcloud, but they are all in the e-43 range and not in the Y8 scale. This is both in rviz and checking manually the intensity values with a script that subscribes to the pointcloud2 topic.
 
Any ideas?
Thanks,
Guido