realsenseai/realsense-ros #1048: rs_multiple_devices filters parameter

danem, 2020-01-07:
It seems that there is no way to specify the filters I wish to activate when using the rs_multiple_devices launch file. For example:

        roslaunch realsense2_camera rs_multiple_devices.launch serial_no_camera1:=<serial1> serial_no_camera2:=<serial2> filters:=pointcloud

The pointcloud topic is never created. Looking through the launch file, it seems like this field is effectively ignored.  Am I missing something?

Thanks

doronhi, 2020-01-09:
You are correct. The field is missing from that launch file.
The launch files are provided as examples of activating different aspects of realsense2_camera and there was never an intention of providing all the fields in every launch file.
You can create your own launch files based on these examples. Specifically, you should add the "filters" field to every node in the rs_multiple_devices.launch, based on examples from rs_camera.launch.

danem, 2020-01-09:
Yeah, that's what I ended up doing. I thought it might have been an oversight worth pointing out.

closed 2020-01-09 by danem