realsenseai/realsense-ros #1049: Set manual exposure prior to streaming

rghl3, 2020-01-08:
Hi,
    I am trying to set manual exposure values for T265 camera using this [launch file.](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/launch/rs_t265.launch)

I have disabled the auto-exposure using below ros param
```
<rosparam>
    camera/tracking_module/enable_auto_exposure: false
</rosparam>
```
as mentioned in [here.](https://github.com/IntelRealSense/librealsense/issues/4413#issuecomment-520200355)

But I am not sure how to set the manual exposure value using launch file.

Adding the below line to rosparam,
```
<rosparam>
    camera/tracking_module/enable_auto_exposure: false
    camera/tracking_module/exposure: 200
</rosparam>
```
produces the following error.
```
To control exposure you must set sensor to manual exposure mode prior to streaming
```
What is the correct procedure to set manual exposure value using launch file?

shannonwarren, 2020-01-30:
@rghl3 Any update on this?

FergalLonerganKUL, 2020-02-05:
@doronhi Could you have a look at this please? Regardless of where you add the rosparam to your launch file the enable_auto_exposure remains set to true when you open rqt_reconfigure with the error regarding setting the exposure before streaming outlined above.

doronhi, 2020-02-10:
It seems like a bug in the librealsense2 library. The enable_auto_exposure option should turn automatically to Off when exposure is set. This bug can be reproduced with the realsense-viewer application.
Thanks for reporting this. We will try to correctly it shortly.

lucadigiammarino, 2020-05-05:
@doronhi is there any update?

closed 2020-10-30 by RealSenseSupport