realsenseai/realsense-ros #1050: Multiple realsense cameras

karry3775, 2020-01-09:
Hello there! I wanted to know if multiple t265 cameras can be launched from ros now. The document online says that t265 doesn't support multiple cameras launch at this moment. Is there a work around this, I really have to use multiple cameras for a project of mine and any help is appreciated. You can really close down this issue as this is not really one. I just wanted to get in touch with you to discuss about the possible solution here. My email: kartikprakash3775@gmail.com

RealSenseSupport, 2020-10-26:
Hi,

Will you be needing further help with this? Unfortunately this ROS functionality is still not available.

If we don’t hear from you in 7 days, this issue will be closed.

Thanks

karry3775, 2020-10-26:
Thank you for your response. You can close this issue. Cheers!

closed 2020-10-26 by RealSenseSupport