realsenseai/realsense-ros #1051: Incorrect Parsing of Physical IDs

samxchandlerpi, 2020-01-10:
I have two realsense d435 cameras connected to a powered USB 3.0 hub, but I can't run them at the same time.

The reason seems to be related to incorrectly parsing the physical IDs of the devices to get the usb port. See the following warning:
`[ WARN] [1578695869.249368440]: Error extracting usb port from device with physical ID: 2-1.1.1.2-28
Please report on github issue at https://github.com/IntelRealSense/realsense-ros`

I also noticed that there was a recent merge to attempt to remedy issues with parsing physical IDs of realsense cameras connected to USB hubs:
#972 

This merge makes the assumption that the physical ID has a format similar to the following:
`[ INFO] [/rear_right/realsense/realsense_camera_manager] [1572376422.393304228]: Device with physical ID /sys/devices/pci0000:00/0000:00:14.0/usb4/4-6/4-6.3/4-6.3:1.0/video4linux/video6 was found.
`

However, my device physical IDs are not being read in that format:
`[ INFO] [1578696525.735386134]: Device with physical ID 2-1.1.1.2-28 was found.`

Here's some more information generated from rs-enumerate-devices:
```
Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	740112070338
    Firmware Version              : 	05.11.10.00
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1.1/2-1.1.1.2/2-1.1.1.2:1.0/video4linux/video1
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Usb Type Descriptor           : 	3.2

 Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	801312071134
    Firmware Version              : 	05.11.10.00
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1.1/2-1.1.1.3/2-1.1.1.3:1.0/video4linux/video4
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Usb Type Descriptor           : 	3.2
```

Finally, it seems like all of the following issues stem from this same problem:
#1018 
#1039 
#1036

doronhi, 2020-01-12:
the message:
`[ WARN] [1578695869.249368440]: Error extracting usb port from device with physical ID: 2-1.1.1.2-28 Please report on github issue at https://github.com/IntelRealSense/realsense-ros`
means that you indeed encountered a usb id that does not match the expected regular expression.
We should change the code accordingly, but in the meantime, the "WARN" header means that you didn't specify the usb port and that the device was picked up by either a serial number or randomly.
Basically it means that the reason you were not able to run 2 devices simultaneously is different.

What I do find intriguing is that the string for "Physical Port" when running `rs-enumerate-devices` is different than the one printed in the message `[ INFO] [1578696525.735386134]: Device with physical ID 2-1.1.1.2-28 was found.`

Could you give more details as to what kind of machine do you run, OS, etc?

samxchandlerpi, 2020-01-12:
> the message:
> `[ WARN] [1578695869.249368440]: Error extracting usb port from device with physical ID: 2-1.1.1.2-28 Please report on github issue at https://github.com/IntelRealSense/realsense-ros` [...]

I'm running Ubuntu 16.04 and ROS Kinetic on a standard x86 architecture based laptop. The only thing out of the norm in my setup could be connecting two realsense cameras to one powered USB 3.0 hub. For various reasons it is difficult for me to change this setup.

Also, I should have clarified in my original post that when I tried to set the usb_port_id I ggot an error instead of the warning. Realsense-ros also asks me to use the serial number instead of the usb_port_id. 

So in my first post my launch file was set up to pass the serial numbers to the realsense-ros nodes and that gave me a warning and, as you said, assigned arbitrary ports to the devices.

If I change the launch file to manually assign the usb_port_ids I get an error that's also related to parsing the physical IDs.

I agree that there seems to be something wrong with the regular expression code and possibly more since the physical ID printed I. The prompt on startup does not match the output of rs-enumerate-devices.

shreyas2311, 2020-01-13:
The same issue is also there on Jetson devices. It doesn't work even when the camera is the only usb that is connected.

samxchandlerpi, 2020-01-13:
Here are some steps I performed to achieve a temporary fix:
1. I plugged in the two realsense devices separately to get the physical IDs in the format expected by the regex parser (as detailed here #972 )
2. I modified the code to statically set the physical IDs based off of the output of rs-enumerate-devices (in line ~65 of realsense_node_factory.cpp)
`	ROS_INFO_STREAM("Device with serial number " << sn << " was found."<<std::endl);
        std::string pn;
        if (strcmp(sn, "740112070338") == 0)
        {
          pn = "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-1/2-1.1/2-1.1.1/2-1.1.1.2/2-1.1.1.2:1.0/video4linux/video4";
        }
        else if (strcmp(sn, "801312071134") == 0)
        {
          pn = "/sys/devices/pci0000:00/0000:00:14.0/usb2/2-3/2-3.1/2-3.1.1/2-3.1.1:1.0/video4linux/video1";
        }
        else
        {
          pn = dev.get_info(RS2_CAMERA_INFO_PHYSICAL_PORT);
        }`
3. I launched the two realsense cameras separately.

I didn't expect to need to perform step 3, but for some reason each node checks all devices which causes the following error:

`ERROR [139715118348032] (handle-libusb.h:87) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY`

shreyas2311, 2020-01-14:
These are steps that I followed:

1) I followed steps from jetsonhacks to patch the kernel - successful.
2) I installed librealsense version 2.28.1 by changing the version number on jetsonhacks script. However, the patch for missing frames issue did not apply properly. However in realsense_viewer, I did not get the missing frames pop-up when I tried the camera with a lower resolution.
3) I installed realsense ros wrapper version 2.2.8 - Build successful.
4) I am able to launch D435 and T265 nodes (I am using a tracking camera as well). In order to launch them, I passed serial_no as a parameter by obtaining them from executing the command rs-enumerate-devices.
Also, I am able to render image and /d400/alighned_depth_to_color topic. However, the depth topic gets stuck very often. This is possibly because the patch for missing frames wasn't successful when I installed librealsense. Also, I keep getting an error saying that "An error has occurred during frame callback".
FInally, I lowered the resolution to 424*280 for color and depth streams. Now, rendering seems much better and I do not get the error very often. However, this isnt ideal as the resolution is very low. Would be great if there a way to apply a proper patch for the missing frames issue.

svrakiss, 2020-01-14:
Could post a link to the patch?.
when I input dmesg i get the following:
[  112.259073] usb usb2: usb_suspend_both: status -16
[  112.277896] usb usb2: usb_suspend_both: status 0
[  112.510681] usb 2-1.3: new SuperSpeed USB device number 3 using tegra-xusb
[  112.531204] usb 2-1.3: New USB device found, idVendor=8086, idProduct=0b07
[  112.531211] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  112.531214] usb 2-1.3: Product: Intel(R) RealSense(TM) Depth Camera 435 
[  112.531218] usb 2-1.3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
[  112.531221] usb 2-1.3: SerialNumber: 933xxxxxxxx
[  112.580102] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[  112.580130] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
[  112.582090] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  112.589896] uvcvideo: Unable to create debugfs 2-3 directory.
[  112.590232] uvcvideo 2-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[  112.600584] uvcvideo 2-1.3:1.0: Entity type for entity Processing 2 was not initialized!
[  112.608760] uvcvideo 2-1.3:1.0: Entity type for entity Camera 1 was not initialized!
[  112.616878] input: Intel(R) RealSense(TM) Depth Ca as /devices/70090000.xusb/usb2/2-1/2-1.3/2-1.3:1.0/input/input5
[  112.618138] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[  112.618150] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[  112.619195] uvcvideo: Unable to create debugfs 2-3 directory.
[  112.619358] uvcvideo 2-1.3:1.3: Entity type for entity Processing 7 was not initialized!
[  112.627528] uvcvideo 2-1.3:1.3: Entity type for entity Extension 8 was not initialized!
[  112.635566] uvcvideo 2-1.3:1.3: Entity type for entity Camera 6 was not initialized!
[  112.643604] usbcore: registered new interface driver uvcvideo
[  112.643607] USB Video Class driver (1.1.1)

I searched online and apparently this can be fixed with a patch to the kernel. But the patch on the realsense git doesn't work with the Jetson Nano.

what's weird is that according to rs-enumerate-devices my physical port is	2-1.3-4.
And the compiled realsense apps work, just not ros.

samxchandlerpi, 2020-01-14:
> Could you post a link to the patch?.
> when I input dmesg i get the following:
> [ 112.259073] usb usb2: usb_suspend_both: status -16
> [ 112.277896] usb usb2: usb_suspend_both: status 0
> [ 112.510681] usb 2-1.3: new SuperSpeed USB device number 3 using tegra-xusb
> [ 112.531204] usb 2-1.3: New USB device found, idVendor=8086, idProduct=0b07
> [ 112.531211] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 112.531214] usb 2-1.3: Product: Intel(R) RealSense(TM) Depth Camera 435
> [ 112.531218] usb 2-1.3: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435
> [ 112.531221] usb 2-1.3: SerialNumber: 933xxxxxxxx
> [ 112.580102] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
> [ 112.580130] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
> [ 112.582090] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435 (8086:0b07)
> [ 112.589896] uvcvideo: Unable to create debugfs 2-3 directory.
> [ 112.590232] uvcvideo 2-1.3:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
> [ 112.600584] uvcvideo 2-1.3:1.0: Entity type for entity Processing 2 was not initialized!
> [ 112.608760] uvcvideo 2-1.3:1.0: Entity type for entity Camera 1 was not initialized!
> [ 112.616878] input: Intel(R) RealSense(TM) Depth Ca as /devices/70090000.xusb/usb2/2-1/2-1.3/2-1.3:1.0/input/input5
> [ 112.618138] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
> [ 112.618150] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435 (8086:0b07)
> [ 112.619195] uvcvideo: Unable to create debugfs 2-3 directory.
> [ 112.619358] uvcvideo 2-1.3:1.3: Entity type for entity Processing 7 was not initialized!
> [ 112.627528] uvcvideo 2-1.3:1.3: Entity type for entity Extension 8 was not initialized!
> [ 112.635566] uvcvideo 2-1.3:1.3: Entity type for entity Camera 6 was not initialized!
> [ 112.643604] usbcore: registered new interface driver uvcvideo
> [ 112.643607] USB Video Class driver (1.1.1)
> 
> I searched online and apparently this can be fixed with a patch to the kernel. But the patch on the realsense git doesn't work with the Jetson Nano.
> 
> what's weird is that according to rs-enumerate-devices my physical port is 2-1.3-4.
> And the compiled realsense apps work, just not ros.

There is no official patch for _this_ issue yet. Also, from your post it seems like you are dealing with _other issues_ related to the Jetson Nano and realsense cameras.

Just follow the steps here and everything should work (I tested this installation guide on the Nano a couple of months ago): https://www.jetsonhacks.com/2019/10/25/realsense-ros-wrapper-jetson-nano/

svrakiss, 2020-01-15:
How do I change the version of realsense-ros? I went and specified 2.2.8 in the install script but when I re ran the examples, it still showed 2.2.12.

shreyas2311, 2020-01-15:
After you clone the realsense-ros repo, cd into it and do git checkout 2.2.8

swinterbotix, 2020-01-16:
I'm currently working with the Intel D435 camera on the Intel NUC6CAYH (running Ubuntu 16.04 with ROS Kinetic). I'm running into a similar issue where 'realsense-viewer' works just fine but realsense-ros gives the 'Error extracting USB port...' message. However, the output of my rs-enumerate devices command is:

[image]

As you can see, the 'Physical Port' message does not give a full path - just part of it. Do you know how this could happen?

FYI - I'm working with Realsense ROS v2.2.12 and LibRealSense v2.31.0.

svrakiss, 2020-01-16:
This is my case, as well. Oddly enough.

samxchandlerpi, 2020-01-16:
> I'm currently working with the Intel D435 camera on the Intel NUC6CAYH (running Ubuntu 16.04 with ROS Kinetic). I'm running into a similar issue where 'realsense-viewer' works just fine but [...]

Can you post the exact error you see when you try to run the realsense node? What is the physical ID of the device that is output by the node on startup?

swinterbotix, 2020-01-16:
[image]

The above is if the Physical Port is defined as 2-3-21 when output from rs-enumerate-devices. I did notice though that if you run the node for the first time after plugging the camera in, sometimes I get errors like...

[image]

... at which point I get notified that the camera will undergo a 'hardware reset'.
Also, despite all this, the node somehow stays alive and I can echo the rostopics, but I'd like to get rid of all these errors and warnings.

samxchandlerpi, 2020-01-17:
> [image]
>
> The above is if the Physical Port is defined as 2-3-21 when output from rs-enumerate-devices. I did notice though that if you run the node for the first time after plugging the camera in, [...]

Okay, other than the fact that rs-enumerate-devices is not printing the full physical port this seems to be the same issue. My setup also works (with those same warnings) if I only use one realsense d435.

I bet that if you try to use two realsense d435s you will start running into problems.

shreyas2311, 2020-01-17:
@svrakiss @swinterbotix I had the same issue as well. ('Error extracting to USB port'). But after I downgraded realsense to 2.28.1 and realsense-ros to 2.2.8, I did not have this issue.

swinterbotix, 2020-01-21:
@svrakiss @shreyas2311, 

I was able to fix my issue without downgrading at all.

All I had to do was type 'sudo apt purge ros-kinetic-librealsense2' into the terminal, hit Enter, then unplug and replug the camera. I saw that the realsense2_camera package depends on librealsense2 but I'm not sure why.... It seems that package was meant to be used with an earlier Linux Kernel - specifically 4.4.X and it could be that somehow that's messing with the installed RealSense Drivers. However, with that package uninstalled, the output of 'rs-enumerate-devices' actually outputs the full string of the Physical Port. Running the RealSense ROS node afterwards then worked perfectly!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@swinterbotix Glad to hear that the issue fixed. Any other questions? Looking forward to your reply. Thanks!

shreyas2311, 2020-02-03:
The issue still persists on the Nvidia Jetson TX2. I have to start the launch files with serial numbers. The output of rs-enumerate-devices however, gives the entire string.

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@shreyas2311 In case we didn't follow the discussion clearly, could you please clarify what issue you're stilling having? Thanks!

doisyg, 2020-02-07:
> @svrakiss @shreyas2311 @doronhi,
> 
> I was able to fix my issue without downgrading at all.
> 
> All I had to do was type 'sudo apt purge ros-kinetic-librealsense2' into the terminal, hit Enter, then unplug and replug the camera. I saw that the realsense2_camera package depends on librealsense2 but I'm not sure why.... It seems that package was meant to be used with an earlier Linux Kernel - specifically 4.4.X and it could be that somehow that's messing with the installed RealSense Drivers. However, with that package uninstalled, the output of 'rs-enumerate-devices' actually outputs the full string of the Physical Port. Running the RealSense ROS node afterwards then worked perfectly!

@svrakiss : thank you so much! I was starting to loose my sanity, I spent a couple of hours trying to understand why I could no launch 2 D435 in the same time (and also port filtering was not working anymore, and the ports naming was messed up, and cameras had to be unpluged/pluged to appear in realsense viewer after a ROS usage) after trying the newly released ros-melodic-realsense2* packages. I still had the issue after uninstalling these packages, however I found out that removing all ros-melodic-realsense2* packages is not uninstalling ros-melodic-librealsense2. After purging it, all went back to normal.

shreyas2311, 2020-02-08:
@RealSenseCustomerSupport The issue is that I am unable to launch D435 camera on the Jetson because of the incorrect parsing of the Port IDs. This issue is not there on the older ros-realsense versions. But in the recent versions, there is a patch where the parsing mechanism has been changed . This is causing errors where the ros-package isn't able to detect the camera properly.

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@shreyas2311 Thanks for the clarification! Could you please specify the detailed SDK and ROS wrapper verion when get the issue? And according to the comment from @swinterbotix https://github.com/IntelRealSense/realsense-ros/issues/1051#issuecomment-576942007 , he/she has resolved the issue. Did you get chance to try his/her sharing? Thanks!

###

gisen-robo-git, 2020-02-19:
> @svrakiss @swinterbotix I had the same issue as well. ('Error extracting USB port...'). But after I downgraded realsense to 2.28.1 and realsense-ros to 2.2.8, I did not have this issue.

I am trying this on ROS kinetic, but realsense-ros requires "ros-kinetic-librealsense2" package and this package only has v2.31.0 on apt repository.
Can I install this package from source ? I think the difference is only the installation path.
I would be grateful if anyone could give me advice.

gisen-robo-git, 2020-02-19:
> I am trying this on ROS kinetic, but realsense-ros requires "ros-kinetic-librealsense2" package and this package only has v2.31.0 on apt repository.
> Can I install this package from source ? I think [...]

I solved this. That was because I installed librealsense2(v2.28.0) from apt repository.
I installed librealsense2 from source, I can build realsense2-camera without "ros-kinetic-librealsense2". It works pretty well in this version.Thank you.

RealSenseCustomerSupport, 2020-02-19:
----------------------------------------------

@shreyas2311 Any update from your side? Thanks!

shreyas2311, 2020-02-19:
@RealSenseCustomerSupport The issue is not there in the librealsense 2.28.1 and the associated realsense-ros. Using these versions, I could get it to work. However, the issue exists in the new ros realsense version 2.30 and 2.31. Therefore, I hope that this issue will be fixed in subsequent versions so that we can use any new features that will be released along with the newer librealsense versions. If realsense has a record of this, we can close the issue here. I initially presumed that the issue will be kept open until a fix is released with a newer librealsense version.

doronhi, 2020-02-20:
The issue of identifying the usb port is, I believe, resolved with #1088.

shreyas2311, 2020-02-20:
That's great. I didn't notice that. We can close the issue here in that case. Thanks!

closed 2020-02-20 by samxchandlerpi