realsenseai/realsense-ros #1052: detached head issue while executing readme

shreyas2311, 2020-01-11:
While executing the following line from the readme, I get an error saying that I am in the detached head state. It would be great if this is resolved soon.

git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`

doronhi, 2020-02-02:
Hi, sorry for the late response.
The message regarding the detached head is not an error. It means that the right version is now on the disk but it is not attached to a specific branch.
Attaching it to a specific branch is optional and is not needed the the successful completion of the building process.
You can attach it to a specific branch of your choice if you want to commit some changes to it, merge it to your own version or any other git option.

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

Hi @shreyas2311,

As @doronhi have said, the detached head state does not interfere with the successful completion of the building process. Please let us know if you need any further clarifications regarding this topic.

Thank you!

shreyas2311, 2020-02-20:
@RealSenseCustomerSupport Alright! Thanks for the update!

closed 2020-02-20 by shreyas2311