realsenseai/realsense-ros #1053: enable_laser2 issue

job-1994, 2020-01-13:
Device: Jetson Xavier
Ubuntu 18.04
ROS Melodic
librealsense 2.31.0
realsense-ros 2.2.12

I am trying to get the streams running using the rs_camera.launch file, setting the initial_reset to true, however it always fails after a few seconds after a Hardware Notification:

[ WARN] [1578911730.399397915]: Hardware Notification:USB REC overflow,1.57891e+12,Error,Hardware Error
[ WARN] [1578911499.052760224]: Hardware Notification:IR stream start failure,1.57891e+12,Error,Hardware Error

leading to 

[ERROR] [1578911499.053030123]: Hardware Reset is needed. use option: initial_reset:=true

For obvious reasons I cannot have the camera resetting every few seconds, and after playing around with a few of the ros parameters for the launch file I have found the following:

when laser1, color, gyro and accel are enabled with the rest set to false, the node runs fine
when laser2 is added to the enabled list, it fails after a few seconds
This even happens when laser2 is enabled on its own. 

Before I dive into the code and such, I was wondering if there were any suggestions that might help me solve this issue, or narrow down where to look for it

RealSenseCustomerSupport, 2020-02-07:
----------------------------------------------

Hi @job-1994,

First of all, could you please provide us with the following details about your system configuration :
HW configuration details (all platforms names, how the camera(s) is/are connected, etc)
Linux Kernel version
number of cameras in active use in the system
for each camera please provide :
   camera model (D435,D435i,etc)
   FW version

Second, please elaborate on "updating the firmware to no avail". What tools do you use?
What's exactly happening when you are trying to update the firmware?

Third, if you reset the camera and run rs-enumerate-devices is Infrared2 still not in
the list of supported modes?

Thank you!

RealSenseCustomerSupport, 2020-02-12:
----------------------------------------------

@job-1994,

Do you still need help with this question?

job-1994, 2020-02-12:
Yes, will try to get you the details tomorrow

RealSenseCustomerSupport, 2020-02-19:
----------------------------------------------

@job-1994,

Did you have a chance to find all the needed details? Do you still need an assistance with this matter?

job-1994, 2020-02-25:
Yes, please see previous post for details

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@job-1994,

We still don't have information about FW version you use and about how the camera is connected to the host system (powered/unpowered hub, direct connection,etc). Please provide us with these details.

Thank you!

P.S.  - please provide answers for the questions we're asking in the new comments instead of
editing your already posted comments.

closed 2020-02-28 by job-1994