realsenseai/realsense-ros #1055: what happens to T265 SLAM when camera sees sparse features(like white simple walls) but still having Intertial and wheel odometry

sadransh, 2020-01-15:
I want to use the tracking camera on a mobile robot with wheel feedback in environments with sparse features.
Can anyone provide me information on whether the T265 slam can keep odometry( to some extent) when it becomes somehow blind(because of sparse features in some scenes it is moving) ? or the odometry will die with only wheel and INS? 

In the case of not working, can anyone suggest other SLAM algorithms designed to survive mentioned conditions?

RealSenseCustomerSupport, 2020-02-12:
----------------------------------------------

Hi @sadransh

It is possible for the T265 SLAM system to work under sparse feature environments.  Although the solution works at its best in environments where features are rich and able to use features to navigate itself in that environment.  Using wheel odometry adds more input into the tracking system and provides another avenue of data into the SLAM fusion.  Being able to specifically say that the system would operate in your specific environment is tough to call but the T265 should be able to function in varying environments.

Whitepapers:  https://dev.intelrealsense.com/docs/whitepapers

Whitepaper on T265:  https://dev.intelrealsense.com/docs/intel-realsensetm-visual-slam-and-the-t265-tracking-camera

schmidtp1, 2020-02-24:
@sadransh, wheel odometry improves the robustness to sparse environments. Yet one issue of the current system could be that it switches to 3DOF (orientation only) mode if there are no features detected at all (e.g. completely white room or completely dark / camera fully(!) covered). Yet I have never seen that in testing (due to the wide FOV).
You can see the effect by running [the wheel odometry example in librealsense](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_wheel_odometry/t265_wheel_odometry.py) and covering the camera partially(!).

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

@sadransh, Let us know if you have any further questions on this topic.

***If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport