realsenseai/realsense-ros #1056: Error extracting usb port and resource temporary unavailable, number: 11

BrooklynBoy21, 2020-01-15:
Hello, I have installed RealSense d435i on Jetson nano, by following the tutorials:
https://www.jetsonhacks.com/2019/05/16/jetson-nano-realsense-depth-camera/

and then install ROS wrapper:
https://www.jetsonhacks.com/2019/10/25/realsense-ros-wrapper-jetson-nano/

I am using L4T 32.3.1 and after launching:
roslaunch realsense2_camera rs_camera.launch

firstly it didn't work at all, after several rebooting it starts working, however I have several problems:

[ WARN] [1579093720.681595033]: Error extracting usb port from device with physical ID: 2-1.3-4
Please report on github issue at https://github.com/IntelRealSense/realsense-ros

and then, periodically I am receiving that error:

 15/01 15:27:24,751 WARNING [546601693568] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

Could someone help me?

FPSychotic, 2020-01-27:
did you tried change the cable?

BrooklynBoy21, 2020-01-28:
@FPSychotic No, I have used the original cable which was in the kit. It is just usb - type C, so I will try to replace it and check.

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@FPSychotic Any luck after replace the USB cable? And what's SDK version and ROS wrapper version when you get this issue?

BrooklynBoy21, 2020-02-11:
@RealSenseCustomerSupport @FPSychotic No, the cable replacement did not help. Camera still work badly with ROS.

I am using ROS Melodic with ROS wrapper 2.2.11
librealsense version: 2.32.1

FPSychotic, 2020-02-11:
No I needed move to other board

RealSenseCustomerSupport, 2020-03-19:
----------------------------------------------

@BrooklynBoy21 The issue should be fixed by https://github.com/IntelRealSense/realsense-ros/pull/1088 which merged into latest release. Could you please have a try? Thanks!

RealSenseCustomerSupport, 2020-03-26:
----------------------------------------------

@BrooklynBoy21 Any update from your side? Thanks!

BrooklynBoy21, 2020-03-26:
@RealSenseCustomerSupport Sorry, I didn't have time to check. As soon as I check it, I will write

jorgemia, 2020-03-26:
My nano is running L4T 32.3.1, I built librealsense from source using JetsonHacksNano and changing the release version to 2.33.1. I also installed realsense-ros 2.2.13 through JetsonHacks.

My D435i is running firmware 5.12.3 and also shows: `15/01 15:27:24,751 WARNING [546601693568] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11` occasionally.

I also sometimes get:
`[547335696832] (uvc-device.cpp:664) interrupt event received: 1, 3, 0, 7, 0, 11, `
along with `IR stream failure` when opening realsense viewer.

RealSenseCustomerSupport, 2020-04-09:
----------------------------------------------

@jorgemia   According to the comment in https://github.com/IntelRealSense/librealsense/issues/6062#issuecomment-604963099 The warning message "control_transfer returned error" reflects communication error with the device that is resolved by repeating the attempt. These messages are still under investigation but as long as they appear only now and then, say at the beginning of the run, they can safely be ignored for now.

RealSenseCustomerSupport, 2020-04-22:
----------------------------------------------

@jorgemia Any other questions about this issue? Thanks!

jorgemia, 2020-04-22:
@RealSenseCustomerSupport No, thanks! Just waiting for a fix.

Fyi, I encounter more issues with `control_transfer returned error` when running the camera multiples times using ROS (The first time I run the camera node the errors only appear at the start and the camera works fine...If I kill the node and later I try to run it again, the camera suffers more problems and sometimes I have to unplug it and plug it in again)

RealSenseCustomerSupport, 2020-04-27:
----------------------------------------------

@jorgemia Could you please create another issue to track "control transfer error" you got? Thanks!
@BrooklynBoy21 Did you get chance to try the latest release to see if the issue resolved? Thanks!

RealSenseCustomerSupport, 2020-05-12:
----------------------------------------------

@BrooklynBoy21 Will close it if no other questions. Thanks!

closed 2020-05-12 by RealSenseCustomerSupport