realsenseai/realsense-ros #1057: USB 2 performance not as advertised

ChrisMOxon, 2020-01-16:
Hi,

We are trying to use the T265 device with USB 2 to just get pose output and in all cases have found significant performance loss to the point that it is unusable.

This has been tried with a Pi 4 as well as multiple desktop PCs, using the realsense viewer app.  In all cases the performance when using USB 2 has been unusable but as soon as the same tests are run on USB 3 we get the advertised <1% error.  

What we have noticed which is strange is that even using USB 2 you can still see the camera output.  According to the docs the camera should only output pose over USB 2 - is this something that needs disabling and could explain the poor performance?

Cheers,
Chris

doisyg, 2020-02-07:
Following this as we are planning to use the T265 in USB2

ChrisMOxon, 2020-02-07:
> Following this as we are planning to use the T265 in USB2

I wouldn't do that if I were you...  I've raised this issue with Intel support directly as we have tested with multiple T265 cameras, in multiple sites and using multiple machines.  Also tested accessing via ROS driver and the Intel Realsense Viewer.  In all cases it's impossible to get any USB2 output worth using.

Intel support's response was that they don't have a camera available for testing with.

doisyg, 2020-02-07:
Surprising as our preliminary tests with the realsense viewer seemed to give us a good tracking.
What type of cable are you using? We did our test with a micro usb 2.0 (smaller, hence not using all the connectors).

schmidtp1, 2020-02-24:
@ChrisMOxon, if I understand correctly you are still streaming images when using USB2. This should be avoided since it is right on the bandwith limit of USB2. Can you try to disable the fisheye streams and retest? Thanks!

RealSenseCustomerSupport, 2020-03-05:
----------------------------------------------

@ChrisMOxon, as mentioned by schmidtp1, if you're connected to USB2 only, only stream the pose data as bandwidth comes into play streaming images and pose, etc.

***If we don’t hear from you in 7 days, this issue will be closed.**

closed 2020-03-17 by RealSenseCustomerSupport