realsenseai/realsense-ros #1058: Improving wheel Odometry API

shannonwarren, 2020-01-16:
I was wondering if it would possible in future releases if we could keep updating the transform between the camera frame and wheel odometer( in future releases ) instead of parsing a constant translation and rotation parameter in the calibration_odometry.json file. I'm currently testing this setup on a two-wheel self-balancing robot with the T265 camera mounted on a rotating head and I observe a lot of odometry drift in z.

Is this in the development pipeline? Thanks.

rgreid, 2020-06-17:
@shannonwarren same situation here. We wrote a quick Python script that queries the `tf` tree and transforms the linear velocity in our `odom` messages from the `base_link` frame to the T265's `rs_t265_pose_frame`.

shannonwarren, 2020-06-19:
@rgreid we do something similar. Do you still face issues of NaN readings?

RealSenseSupport, 2020-10-26:
Hi @shannonwarren,

Thanks for this enhancement request. At this time we have moved our focus to our next generation of products and consequentially will not be addressing any new T265 features.

Thanks

closed 2020-10-26 by RealSenseSupport