realsenseai/realsense-ros #1060: align_depth doesn't work

95ImadEL, 2020-01-20:
Hello everyone,

I have a D435 realsense camera, I installed the latest release of realsense-ros and realsense SDK, and I'd like to have aligned depth to infrared camera, so I set align_depth parameter to true in my launch file. The problem is that i receive no new messages on the topic /camera_d435/aligned_depth_to_infra1/image_raw
Could you help me to solve this issue ? Thank you in advance.

[image]

doronhi, 2020-01-30:
What version and kernel do you use?
This issue could be related to problems in timestamps as described in #1047

RealSenseCustomerSupport, 2020-05-19:
----------------------------------------------

Hi.

Will you be needing further help with this?

The topic is publishing as expected on a D435 as of RealSense ROS v2.2.13 and LibRealsense v2.33.1

If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-06-03 by RealSenseCustomerSupport