realsenseai/realsense-ros #1061: Odometry jumps even when enable_pose_jumping is set to false

shannonwarren, 2020-01-20:
I've noticed the output of camera odometry from the T265 jumps (indoor) even when options of enable_pose_jumping, enable_mapping and enable_relocalization are set to false. I'm using the latest librealsense version with wheel odometry integrated.

radfordi, 2020-01-28:
The pose shouldn't jump.   I wonder if these values are making it down the to T265?  Would you post your launch file for completeness?

shannonwarren, 2020-02-06:
@radfordi Thanks for the response. You were right about using the options, I used rqt_reconfigure and found that there was an error in my namespace. Besides this, I have seen the NaN issue as raised in https://github.com/IntelRealSense/librealsense/issues/4518 and https://github.com/IntelRealSense/realsense-ros/issues/955 occurring quite frequently after the robot patrols for long periods and then suddenly halts ( will attach a bag file ).

closed 2020-05-12 by shannonwarren