realsenseai/realsense-ros #1062: cmake .. error (RTABmap for raspbian)

CluelessIT, 2020-01-21:
Hi guys,

I'm pretty new to this system and working on a project to generate a 3D map using intel realsense D435i with Rpi 4(Raspbian Buster). 

While installing RTABmap I met this weird error when i input cmake ..
CMake Deprecation Warning at CMakeLists.txt:64 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:67 (cmake_policy):
  The OLD behavior for policy CMP0043 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Error at CMakeLists.txt:196 (FIND_PACKAGE):
  By not providing "FindOpenCV.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "OpenCV", but
  CMake did not find one.

  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/pi/rtabmap/build/CMakeFiles/CMakeOutput.log".

Can anyone advise me on what I did wrong? thank you!

FPSychotic, 2020-01-23:
rtabmap has excellent manual and forum on their website, even if you fix that error, you will receive others different. You need decide what installation do you want, if standalone or ros dependent.So I would recommend you go to the official site and forum.

CluelessIT, 2020-01-24:
Alright! Let me try and see, thanks! @FPSychotic

CluelessIT, 2020-01-27:
@FPSychotic Hi, I've tried the standalone installation initially but it can't seem to work.

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@CluelessIT What's the issue details now? Could you please update? Thanks!

RealSenseCustomerSupport, 2020-02-17:
----------------------------------------------

@CluelessIT Could you please update the issue details? Looking forward to your update. Thanks!

closed 2020-02-25 by CluelessIT