realsenseai/realsense-ros #1063: Lost transform between camera_imu_optical_frame and camera_link

devanshbatra04, 2020-01-23:
I am trying to accomplish indoor mapping using RTAB-MAP with a D435i as documented in https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i
I have been able to do that successfully before, but I am running into this error now. 

I had tried to compile Rtabmap-ros from scratch, which may have been an issue.
I tried uninstalling the scratch version using make uninstall and using the apt version, but it hasn't helped.
[image]

I am on a deadline here, so any help would be useful. It is probably not a Rtabmap issue.

RealSenseCustomerSupport, 2020-02-21:
----------------------------------------------

Hi devanshbatra04,

Are you still seeing this? Can you try incorporating this PR into your build. It's part of a future released but contains fixes that might resolve your issue.

https://github.com/IntelRealSense/librealsense/pull/5751

Thanks

RealSenseCustomerSupport, 2020-04-06:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-04-13 by RealSenseCustomerSupport