realsenseai/realsense-ros #1064: use_nominal_extrinsics in xacro

fherreroc, 2020-01-24:
In `realsense2_description/urdf/_d435.urdf.xacro`, a use_nominal_extrinsics xacro:macro parameter is defined (`use_nominal_extrinsics:=false`) but it is not used, as later on line 75, a xacro argument is used instead (<xacro:if value="$(arg use_nominal_extrinsics)">).

If it is intended to be used as an argument, the xacro parameter is useless and can cause confusion, so it could be removed.

RealSenseCustomerSupport, 2020-02-24:
----------------------------------------------

Hi fherreroc,

Actually it's used from the launch file here...

https://github.com/IntelRealSense/realsense-ros/blob/69d199d9e01a7c57b78831a6486c83ad8653aba9/realsense2_description/launch/view_d435_model.launch#L2

Here's the explanation from the original PR:

https://github.com/IntelRealSense/realsense-ros/pull/886

Thanks

fherreroc, 2020-02-25:
Yes, in that launch file the **argument** is passed/used, but not the xacro **parameter** (which is defined but not used within _d435.urdf.xacro)
That's what I mean with "duality", it is defined in two ways, but only one is actually used. 
Also, the argument has no default value, so if you do not define it in your own launch file or xacro, you get the error "undefined substitution argument use_nominal_extrinsics None"

mcamurri, 2020-03-14:
I noticed the same problem

mcamurri, 2020-03-15:
I've made a PR with the fix: #1125

RealSenseSupport, 2020-08-14:
Hi @mcamurri,

Would you say we can close this now? 

FYI if we don’t hear back in 7 days, this issue will be closed.

Thanks again for your contribution!