realsenseai/realsense-ros #1065: Failed launch of rs_camera D415

chrispduck, 2020-01-27:
I am having issues launching rs_camera.launch. Once launched, the relevant rostopics (mainly depth) has no data on it. I'm using a D415 with ROS kinetic.

Any help would be appreciated.

```
user@tullus:~/catkin_ws/src/realsense-ros/realsense2_camera$ roslaunch realsense2_camera rs_camera.launch 
... logging to /home/user/.ros/log/2855d00c-4119-11ea-be26-00249b27e4c8/roslaunch-tullus-16338.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server <ip>

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 71 lines cut ...]
[ INFO] [1580138690.478148369]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ WARN] [1580138690.478188459]: Given stream configuration is not supported by the device!  Stream: Infrared, Stream Index: 2, Width: 640, Height: 480, FPS: 30
[ INFO] [1580138690.478949313]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1580138690.478971942]: setupPublishers...
[ INFO] [1580138690.481737949]: Expected frequency for depth = 30.00000
[ INFO] [1580138690.508091914]: Expected frequency for infra1 = 30.00000
[ INFO] [1580138690.534894333]: Expected frequency for color = 30.00000
[ INFO] [1580138690.562504330]: setupStreams...
[ INFO] [1580138690.563431756]: insert Depth to Stereo Module
[ INFO] [1580138690.563474988]: insert Color to RGB Camera
[ INFO] [1580138690.563495033]: insert Infrared to Stereo Module
[ INFO] [1580138690.623278967]: SELECTED BASE:Depth, 0
[ INFO] [1580138690.633082285]: RealSense Node Is Up!
 27/01 15:24:50,802 WARNING [140036510926592] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1580138690.803156885]: Frame metadata isn't available! (frame_timestamp_domain = RS2_TIMESTAMP_DOMAIN_SYSTEM_TIME)
 27/01 15:24:54,908 WARNING [140036529575680] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1580138695.492030019]: Hardware Notification:REC error,1.58014e+12,Error,Hardware Error
[ WARN] [1580138697.493065280]: Hardware Notification:USB SCP overflow,1.58014e+12,Error,Hardware Error
[ WARN] [1580138701.505508121]: Hardware Notification:USB SCP overflow,1.58014e+12,Error,Hardware Error
[ WARN] [1580138702.506022294]: Hardware Notification:USB CAM overflow,1.58014e+12,Error,Hardware Error
[ WARN] [1580138703.515666722]: Hardware Notification:USB SCP overflow,1.58014e+12,Error,Hardware Error
[ WARN] [1580138704.516326166]: Hardware Notification:USB CAM overflow,1.58014e+12,Error,Hardware Error
[ WARN] [1580138705.517040226]: Hardware Notification:USB SCP overflow,1.58014e+12,Error,Hardware Error

```

doronhi, 2020-01-30:
What machine are you running on? Are there any other heavy processes in the background?
The warning messages suggests the machine can't keep up.

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

Hi @chrispduck,

Do you still need help with this question?

chrispduck, 2020-02-16:
It seems as though the machine was not capable of the high bandwidth required by the camera. It only had USB 2.0 ports. On another machine with USB 3.0 support, it works fine! 

Feel free to close the issue.
Cheers

closed 2020-02-16 by chrispduck