realsenseai/realsense-ros #1068: D435i rs_camera Error: "Call to publish() on an invalid Publisher"

boewing, 2020-01-29:
I have a D435i camera running on a Jetson Nano with Jetpack 4.2.3. My task is to publish pointclouds to my ROS network (ROS melodic). 

Unfortunately, in ca. 40 % of the time the rs_camera.launch node with default parameters does not start due to an error:
```
[FATAL] [1580322128.643459713]: ASSERTION FAILED
	file = /opt/ros/melodic/include/ros/publisher.h
	line = 102
	cond = false
	message = 
[FATAL] [1580322128.643593257]: Call to publish() on an invalid Publisher
[FATAL] [1580322128.643640757]: 
```

 (Full Logfile here: [errors.txt](https://github.com/IntelRealSense/realsense-ros/files/4130030/errors.txt))

The speciality in my setup is that my roscore runs on a laptop connected over wifi. However, when I run the roscore on the Jetson Nano (which is running rs_camera.launch) then it does not have an error. (Unfortunately, in my setup roscore cannot run on Jetson Nano)

I guess there is a race condition somewhere in realsense-ros: When something wants to be published before the publisher was initialized, I get an error. 

How can this problem be solved? Where do I need to check if the publisher is valid?

jonathanhechtbauer-bc, 2020-01-30:
same problem after pulling latest changes from this repo

armandok, 2020-02-26:
I am having the same issue on nano with rs_camera and rs_aligned_depth.

doronhi, 2020-03-01:
This sounds like a serious error. I will try to attend to it shortly.
In the meanwhile, creation of publishers is done in the [void BaseRealSenseNode::setupPublishers()](https://github.com/IntelRealSense/realsense-ros/blob/4d565649a73fad8dbedc79ef59dd1d5baeca613c/realsense2_camera/src/base_realsense_node.cpp#L770) function. I hope it is a good starting point for investigation.

RealSenseSupport, 2020-12-09:
Hi,

Are you still experiencing this on latest? Will you be needing further help? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

tcies, 2020-12-10:
Hi, I was supervising the project by @boewing and am still planning to work with this setup in the near future. From our side, I think this issue can be closed, for now, I will reopen it if I encounter the issue again. Thank you very much!

closed 2021-01-15 by RealSenseSupport