realsenseai/realsense-ros #1069: Nothing published when using an external nodelet manager

mjsobrep, 2020-01-30:
This might be similar to #697 , but I am still having trouble. 

If I run this launch file (case 1):
```xml
<launch>
    <node pkg="nodelet" type="nodelet" name="cust_manager"  args="manager"/>
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="external_manager" value="false"/>
        <arg name="manager"          value="cust_manager"/>
    </include>
</launch>
```
Then things work well enough for me, some video gets published, etc. 

If I run this though (case 2):
```xml
<launch>
    <node pkg="nodelet" type="nodelet" name="cust_manager"  args="manager"/>
    <include file="$(find realsense2_camera)/launch/rs_camera.launch">
        <arg name="external_manager" value="true"/>
        <arg name="manager"          value="cust_manager"/>
    </include>
</launch>
```
Then nothing gets published. 

Any thoughts as to what is going on?

## Case 1 launch printouts:
<pre> roslaunch flo_telepresence min_realsense.launch 
... logging to /home/nuc-admin/.ros/log/638b85a0-43a2-11ea-b0d9-e470b83e61ec/roslaunch-flo-nuc-4060.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is &lt;1GB.

<b>started roslaunch server http://10.42.0.189:44275/</b>

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
[... 250 lines cut ...]
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /camera/
    realsense2_camera (nodelet/nodelet)
  /
    cust_manager (nodelet/nodelet)

auto-starting new master
<b>process[master]: started with pid [5538]</b>
<b>ROS_MASTER_URI=http://localhost:11311</b>

<b>setting /run_id to f41e642a-43a2-11ea-b0d9-e470b83e61ec</b>
<b>process[rosout-1]: started with pid [5551]</b>
started core service [/rosout]
<b>process[cust_manager-2]: started with pid [5568]</b>
<b>process[camera/realsense2_camera-3]: started with pid [5569]</b>
</pre>

## Case 2 topics:
<pre><font color="#4E9A06">rostopic</font> list                      
/rosout
/rosout_agg
</pre>

RealSenseCustomerSupport, 2020-04-07:
----------------------------------------------

Hi @mjsobrep,

An ability to use external manager has been added long time ago. Can you please clarify what are you trying to achieve in both cases and describe the structure of your system including nodes/nodelets that should run in the system, managers, etc.

Thank you!

RealSenseCustomerSupport, 2020-04-11:
----------------------------------------------

@mjsobrep,

Do you still need help with this question?

closed 2020-04-20 by RealSenseCustomerSupport

mjsobrep, 2020-05-11:
@RealSenseCustomerSupport Sorry I was slow to respond. In the 2 months between my original question and your response I had to move on. I might come back to this later...

shuhaowu, 2020-10-24:
I ran into the same issue. You need to have the absolute path in `external_manager`, like this:

```
 <arg name="manager"          value="/cust_manager"/>
```

Otherwise, it tries to find cust_manager in the camera namespace.