realsenseai/realsense-ros #1072: Frame rate can't get more of 15fps

FPSychotic, 2020-02-04:
I have my d435 upgraded to the last firmware.
I have had issues on every SBC or PC can get work together d435 and t265 simultaneously (Arm64 and X86 SBC and X86 PC nuc like) which I will detail on other post. After find work arounds for every issue, result that I detected than in no one system I have, D435 is able to go higher of 15hz when I make rostopic hz to any node. I tested on 4.15, 4.19,5.x kernel, on raspbian, ubuntu 18.04, arm64,.x86, librealsense 2.28.1 and 2.31, with original lauch files and modified, often at qvga resolutions, the nodes when are launched tell I should expect 30fs, the resolution modes are supported at 30fps as is written on rs-enumerate-devices, I  used different cables, from r200, original, and others, do not change anything run it on USB2 or USB3 mode, with or without t265 connected. No difference from patched kernel to uvc video backend instalation, from source or apt with dkms
It looks that the new firmware 5.12.02.100 is not good.
Just tell that on ARM64 Raspberry 4, And Jetson Nano cant run both cameras, same on 2x UP boards with different kernels and an i7 u8565, on all of them resource busy, USB physical address conflict, that t265 do not connect after boot, that t265 must be connect before...
Here there is a good mess, as on 6 different boards had the same errors depensing the SDK version.

mcamurri, 2020-02-13:
I'm also facing performance drop with the latest releases.
I'm testing a RealSense D435i on a DELL Precision 7530 laptop running Ubuntu 18.04 and ROS melodic, both upgraded to latest .

`uname -r`:
 `4.18.0-17-generic` 

`lswh -short`:
```
H/W path       Device    Class          Description
===================================================
                         system         Precision 7530 (0831)
/0                       bus            0C1D71
/0/0                     memory         64KiB BIOS
/0/42                    memory         16GiB System Memory
/0/42/0                  memory         8GiB SODIMM DDR4 Synchronous 2667 MHz (0.4 ns)
/0/42/1                  memory         DIMMProject-Id-Version: lshwReport-Msgid-Bugs-To: FULL NAME <EMAIL@ADDRESS>PO-Revision-Date: 2012-02-05 00:26+0000Last-Translator: Andi Chandler <Unknown>Language-Team: English (United Kingdom) <en_GB@li.org>MIME-Version: 1.0Content-Type: text/plain; charset=UTF-8Content-Transfer-Encoding: 8bitX-Launchpad-Export-Date: 2018-07-12 13:19+0000X-Generator: Launchpad (build 18719) [empty]
/0/42/2                  memory         8GiB SODIMM DDR4 Synchronous 2667 MHz (0.4 ns)
/0/42/3                  memory         DIMMProject-Id-Version: lshwReport-Msgid-Bugs-To: FULL NAME <EMAIL@ADDRESS>PO-Revision-Date: 2012-02-05 00:26+0000Last-Translator: Andi Chandler <Unknown>Language-Team: English (United Kingdom) <en_GB@li.org>MIME-Version: 1.0Content-Type: text/plain; charset=UTF-8Content-Transfer-Encoding: 8bitX-Launchpad-Export-Date: 2018-07-12 13:19+0000X-Generator: Launchpad (build 18719) [empty]
/0/4d                    memory         384KiB L1 cache
/0/4e                    memory         1536KiB L2 cache
/0/4f                    memory         12MiB L3 cache
/0/50                    processor      Intel(R) Xeon(R) E-2186M  CPU @ 2.90GHz
/0/100                   bridge         8th Gen Core Processor Host Bridge/DRAM Registers
/0/100/1                 bridge         Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16)
/0/100/1/0               display        GP107GLM [Quadro P2000 Mobile]
/0/100/1/0.1             multimedia     GP107GL High Definition Audio Controller
/0/100/4                 generic        Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem
/0/100/8                 generic        Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th Gen Core Processor Gaussian Mixture Model
/0/100/12                generic        Cannon Lake PCH Thermal Controller
/0/100/14                bus            Cannon Lake PCH USB 3.1 xHCI Host Controller
/0/100/14/0    usb1      bus            xHCI Host Controller
/0/100/14/0/a            generic        5880
/0/100/14/0/b            multimedia     Integrated_Webcam_HD
/0/100/14/0/e            communication  Bluetooth wireless interface
/0/100/14/1    usb2      bus            xHCI Host Controller
/0/100/14/1/1            multimedia     Intel(R) RealSense(TM) Depth Camera 435i
/0/100/14.2              memory         RAM memory
/0/100/15                bus            Intel Corporation
/0/100/15.1              bus            Intel Corporation
/0/100/16                communication  Cannon Lake PCH HECI Controller
/0/100/17                storage        Intel Corporation
/0/100/1b                bridge         Cannon Lake PCH PCI Express Root Port 21
/0/100/1b/0              storage        SK hynix
/0/100/1c                bridge         Intel Corporation
/0/100/1c.5              bridge         Intel Corporation
/0/100/1c.5/0            generic        Realtek Semiconductor Co., Ltd.
/0/100/1c.6              bridge         Intel Corporation
/0/100/1c.6/0  wlp111s0  network        Wireless-AC 9260
/0/100/1f                bridge         Intel Corporation
/0/100/1f.3              multimedia     Cannon Lake PCH cAVS
/0/100/1f.4              bus            Cannon Lake PCH SMBus Controller
/0/100/1f.5              bus            Cannon Lake PCH SPI Controller
/0/100/1f.6    eno1      network        Ethernet Connection (7) I219-LM
/1                       power          DELL GW0K98A
```
I've followed all the instructions in your readme, installing the packages from APT:
[image]
 and cloning this repository  v2.2.12, catkin compiled in `Release` mode

Finally, I've upgraded the firmware to version `05.12.02.100`

When I run `realsense-viewer` I see lots of errors about incomplete frames:
[image]

When I run the ROS wrapper I get very low framerates and warnings about USB buffer:
```
rostopic hz /camera/infra1/image_rect_raw 
subscribed to [/camera/infra1/image_rect_raw]
average rate: 5.989
	min: 0.100s max: 0.267s std dev: 0.06307s window: 6
[... 149 lines cut ...]
[ INFO] [1581626089.188173760]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1581626089.188725280]: infra1 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1581626089.189262169]: infra2 stream is enabled - width: 640, height: 480, fps: 30, Format: Y8
[ INFO] [1581626089.207843946]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1581626089.208112962]: setupPublishers...
[ INFO] [1581626089.211718972]: Expected frequency for depth = 30.00000
[ INFO] [1581626089.215521462]: Expected frequency for infra1 = 30.00000
[ INFO] [1581626089.217754212]: Expected frequency for infra2 = 30.00000
[ INFO] [1581626089.219855068]: Expected frequency for color = 30.00000
[ INFO] [1581626089.225583738]: setupStreams...
[... cut ...]
[ WARN] [1581626092.181855705]: Hardware Notification:USB SCP overflow,1.58163e+12,Error,Hardware Error
[ WARN] [1581626094.182480173]: Hardware Notification:USB SCP overflow,1.58163e+12,Error,Hardware Error
[ WARN] [1581626096.183477945]: Hardware Notification:USB SCP overflow,1.58163e+12,Error,Hardware Error
[ WARN] [1581626098.184177696]: Hardware Notification:USB SCP overflow,1.58163e+12,Error,Hardware Error
```

Finally, the new firmware has the weird side effect of rotating the screen of my laptop whenever the camera is placed upside-down...

YoshuaNava, 2020-02-21:
@mcamurri I have also experience the rotating screen issue with D435i cameras, and I think that it is related to the Noveau driver in some way. It doesn't happen when I switch to a proprietary GPU driver.

Regarding the issues that you experience, we have not observed that behavior with the kernel patch, kernel 4.15 and the latest software/firmware. Quite the opposite. I do see that your camera is recognized by your PC as a USB 3.2 device, which is OK. 

Can you check that your laptop doesn't have Secure Boot enabled? Or any limitation on the USB bandwidth set in the BIOS?

mcamurri, 2020-02-21:
No secure boot or limitation. In fact, with this configuration I get the nominal 30 Hz rate without changing any bios settings:
- 4.15.0-72-generic
- librealsense 2.21.0 (compiled manually)
- realsense-ros 2.2.3 (compiled manually)
- firmware 05.10.13.00

The rotation happens even with the proprietary graphics drivers (firmware used is 05.11.01.100 in this case):
[image]

But that's a minor problem.

I'll try again the whole installation procedure in a few weeks.

RealSenseSupport, 2020-12-05:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

mcamurri, 2020-12-07:
All is good with the most recent kernel, firmware & drivers. Thanks.

closed 2020-12-14 by RealSenseSupport