realsenseai/realsense-ros #1073: role of catkin package librealsense2

christian-rauch, 2020-02-05:
As far as I understand the [`librealsense`](https://github.com/IntelRealSense/librealsense) provides the kernel-space and user-space driver. Both need to be installed to build and run the ROS node.

However, it is unclear to me, which role the catkin package for the driver (i.e. `ros-melodic-librealsense2`) plays. It seems to only provide the user-space component to allow building the ROS node, but it will not be sufficient to run the ROS node since it does not contain the kernel driver.

What is the purpose of [`ros-melodic-librealsense2`](https://github.com/ros/rosdistro/blob/d973df16dfeb852c7762b1e31f0ae54cd5793bcd/melodic/distribution.yaml#L4436-L4451) and why should it be installed?

doronhi, 2020-02-10:
realsense2_camera, the ROS wrapper for librealsense, obviously depends on librealsense2 library.
librealsense2 library is, as for now, not available in public servers but only here: http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo
As a consequence, `rosdep install` fails when encountering this dependency.
The temporary solution was to add librealsense2 to the ROS public server. This package is named: ros-melodic-librealsense2.

ros-melodic-librealsense2 build differs from that of the package librealsense2 in 2 aspects:
1.	To avoid the x11 dependency, ros-melodic-librealsense2 does not include examples and tools
2.	To avoid the kernel patches dependency, ros-melodic-librealsense2 is built to use libuvc driver and not the native v4l driver.

The difference between using libuvc driver and v4l is better explained [here]( https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604).
In short, v4l driver works in kernel space and libuvc works in user space.
The native kernel's v4l driver does not yet support all the additional data that the device produces for every frame. To gain access to all the data one must update the kernel. One option is to install librealsense2-dkms.
However, kernel patches are not always an option. Such is the case when dealing with Nvidia's versions of ubuntu or other Linux distributions.
An alternative option is to use librealsense libuvc backend which can interpret all the additional frame's data without kernel modifications. It has some limitations, some are still being debugged, but they do not require the kernel patches installation and should work with all Linux distributions.

Note that depending on the kernel version you use, you may have all the additional frames' data that **you** need using v4l driver even without the kernel patches.

christian-rauch, 2020-02-10:
Thank you for the details. That was helpful to understand the differences.

So from my understanding, the ROS wrapper is using the user-space driver (with `uvc` backend) and there is no advantage from installing the `librealsense2-dkms` (the kernel-space driver with `v4l` backend) when using ROS only.

Is there a difference in performance (CPU usage, maximum FPS) between the `uvc` and `v4l` backend? Why would you use the native `v4l` that runs in kernel space, but has less functionality than the user-space `uvc` driver?

In the long-term do you plan to support only one backend?

doronhi, 2020-02-10:
The ROS wrapper use whatever driver the librealsense2 uses.
There are benefits to using the kernel based v4l driver as mentioned in the discussion in the link I provided, including stability and performance. I didn't compare and I would guess the performance difference will depend on the machine and the workload at hand.
The kernel based driver can close the device in an orderly fashion if the application crash and can have better management of multiple devices. The v4l driver is part of the widely spread Linux kernel and benefits from a large opensource community.

In the long term we hope the kernel patches will be integrated into the official Linux kernel. Libuvc will be supported for there always will be platforms that need it, such as Android, Mac etc.

closed 2020-08-16 by christian-rauch