realsenseai/realsense-ros #1074: New LibrealSense 2.32.0 not supported by the current Realsense-ros package?

shannonwarren, 2020-02-08:


MartyG-RealSense, 2020-02-08:
The RealSense ROS wrapper is not in sync with the release schedule of the RealSense SDK.  At the time of writing this, the ROS wrapper's most recent release supports 2.31.

https://github.com/IntelRealSense/realsense-ros

shannonwarren, 2020-02-08:
Thanks @MartyG-RealSense. As mentioned in the release notes https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2321 there have been improvements with respect to NaN readings which I have been encountering often https://github.com/IntelRealSense/realsense-ros/issues/1061.

MartyG-RealSense, 2020-02-08:
Typically, updates to the wrapper arrive on an unpredictable schedule.  I'll tag the wrapper developer @doronhi into this conversation to see if they can provide an official estimate of when the next wrapper version will arrive.

doronhi, 2020-02-09:
There are bug fixes in librealsense but as for now, no major improvement has entered the wrapper, therefor no new release was published. The Current version, to the best of my knowledge, supports librealsense v2.32.1.

shannonwarren, 2020-02-11:
@doronhi the readme says the current supported version of librealsense is v2.31.0.

closed 2020-02-12 by shannonwarren