realsenseai/realsense-ros #1076: cannot record pointcloud with rosbag record -a using demo_pointcloud.launch

sadransh, 2020-02-11:
OS : Ubuntu 16.04 
ROS : kinetic 
librealsense : v 2.32.1 
device : d435i 
fw version : 05.12.02.100

I am trying to record and replay rosbag using `rosbag record -a` with `demo_pointcloud.launch`
( experienced similar issue trying with rs_d400_t265.launch file) 

this is the output : 
```

`[ INFO] [1581391365.284728560]: RealSense Node Is Up!
[ERROR] [1581391381.141559412]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1581391381.167621097]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391381.321075826]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1581391381.645924984]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1581391381.647696160]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391381.686697173]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391381.708229145]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391381.728427843]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391381.750428763]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391381.791974533]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1581391381.817293897]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391381.884029996]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391381.951158192]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.017960187]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.084901159]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.151838548]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.218809505]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.250077963]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1581391382.285749669]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.352695919]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.419677624]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.486719595]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.553637005]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.620506941]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.687601175]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.720678113]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1581391382.754439819]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.821666005]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.888488449]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391382.955312711]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.022408950]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.089230181]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.156180487]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.177172643]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ WARN] [1581391383.179770297]: No stream match for pointcloud chosen texture Process - Color
[ERROR] [1581391383.223454502]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.290274279]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.357116945]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.423952782]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.491017252]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.558065283]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.625134894]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.632464173]: cv_bridge exception: '[16UC1] is not a color format. but [bgr8] is. The conversion does not make sense'
[ERROR] [1581391383.691944016]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.758786651]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.825962509]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.892874413]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).
[ERROR] [1581391383.959763424]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: rgb8).`
```

I have also tried removing some topics from recoding and recording only the list below : 
```
/camera/color/camera_info
[... cut ...]
/camera/depth/image_rect_raw/compressedDepth
/camera/depth/image_rect_raw/theora
/camera/extrinsics/depth_to_color
/tf
/tf_static
```
removing following topics stops the error, however, the recorded rosbag will not show any pointcloud using pointcloud2 in rviz(on points topic) : 
```
/camera/color/image_raw/theora
/camera/depth/image_rect_raw/theora
/camera/depth/image_rect_raw/compressedDepth
/camera/color/image_raw/compressedDepth
```
Can anyone tell how is it possible to record pointclouds and other topics correctly?

doronhi, 2020-02-16:
There are issues recording compressed depth. See [here](https://github.com/ros-drivers/usb_cam/issues/55)

Indeed I got a lot of errors trying to record the list of topics you describe.
Running the followings:
`rosbag record /camera/color/camera_info /camera/color/image_raw /camera/depth/camera_info /camera/depth/color/points /camera/depth/image_rect_raw /camera/extrinsics/depth_to_color /tf /tf_static -O temp_bag.bag`
and then play the recorded file:
`rosbag play temp_bag.bag`
I can see the pointclound in rviz on topic `/camera/depth/color/points`

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@sadransh  Any other questions with doronhi's reply? Looking forward to your update. Thanks!

sadransh, 2020-02-26:
No that was all. Thanks for fast reply. Everything worked fine. I will reopen the issue in case of any related issue.

closed 2020-02-26 by sadransh

golnazhabibi3, 2020-08-14:
Thanks, that works, but it occupies a huge space (10 GB for 2-3 minutes), is there anyway that we can record the compressed image/depth? thanks!

doronhi, 2020-08-14:
You can record the topic : `rosbag record /camera/color/image_raw/compressed`
If you don't see this topic you may need to install the plugin: `sudo apt install ros-kinetic-compressed-image-transport`
You can change the level of compression at the parameter: "/camera/color/image_raw/compressed/png_level" using rqt_reconfigure.
Same goes for depth images as well.