realsenseai/realsense-ros #1077: pointcloud is occasionally not published

freedomtrain, 2020-02-12:
I have installed realsense-ros and now trying to display pointcloud on rviz by using command roslaunch realsense2_camera rs_camera.launch filters:=pointcloud. The rgb, depth, and infrared worked just fine.

Here is my environment:
OS: Ubuntu 18.04
ROS: Melodic
Librealsense: v2.32.1
Realsense ROS: v2.2.12
Device: D435

The problem is when I look at /camera/depth/color/points it doesn't publish any data at all resulted to the no pointcloud visualization in rviz. I tried to reset the calibration settings from [this](https://github.com/IntelRealSense/realsense-ros/issues/595) but had no luck. I tried to reinstall the firmware and somehow the pointcloud is published. Then I closed and relaunched the ros session and the pointcloud didn't work again. I tried the same solution as before but it still didn't work!! I tried to restart ubuntu and redo the solution but still no luck. A few days later when I started the ubuntu again it magically worked! I closed the ros session again and it back again, won't published any topic! This really drives me crazy.

Here is the log when the pointcloud is working

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: pointcloud
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
[... 203 lines cut ...]
[ INFO] [1580920471.913024702]: RGB Camera was found.
[ INFO] [1580920471.913045755]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580920471.913063123]: (Fisheye, 1) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580920471.913084499]: (Fisheye, 2) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580920471.913102088]: (Gyro, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580920471.913115738]: (Accel, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580920471.913134226]: (Pose, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1580920471.913165788]: Add Filter: pointcloud
[ INFO] [1580920471.916892714]: num_filters: 1
[ INFO] [1580920471.916920502]: Setting Dynamic reconfig parameters.
[ INFO] [1580920471.951909992]: Done Setting Dynamic reconfig parameters.
[ INFO] [1580920471.957052555]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1580920471.971206067]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1580920471.971271308]: setupPublishers...
[ INFO] [1580920471.973231604]: Expected frequency for depth = 30.00000
[ INFO] [1580920471.994721075]: Expected frequency for color = 30.00000
[ INFO] [1580920472.009111779]: setupStreams...
[ INFO] [1580920472.010068378]: insert Depth to Stereo Module
[ INFO] [1580920472.010231860]: insert Color to RGB Camera
[ INFO] [1580920472.024198479]: SELECTED BASE:Depth, 0
[ INFO] [1580920472.032094065]: RealSense Node Is Up!
 05/02 23:34:32,196 WARNING [140158039406336] (ds5-timestamp.cpp:76) UVC metadata payloads not available. Please refer to the installation chapter for details.
[ WARN] [1580920479.805877560]: No stream match for pointcloud chosen texture Process - Color

Note that the latter gives a [WARN] about no stream match for pointcloud but I don't really understand what this is. Does anyone have suggestions? Thanks in advance!

doronhi, 2020-02-16:
It sound very very frustrating indeed and I am sorry for that.
You didn't specify what kernel do you use but it sounds like it could be related to time stamp issues that were fixed [here](https://github.com/IntelRealSense/librealsense/pull/5751).
The fix is not yet published as a debian package (it will be in a week or two) so if in a hurry you'll have to build from source.

freedomtrain, 2020-02-18:
Thanks for your response. My kernel version is 5.0.0-37-generic.
I think I have solved the problem by enabling pointcloud_texture_stream:=RS2_STREAM_ANY. The point cloud isn't RGB-textured but it doesnt matter for me.

However, I encountered similar problem. Any of the aligned_to_depth topics won't show up. It is also present when point cloud won't publish. When I enabled the RS2_STREAM_ANY the problem still persists. Could it be related to time stamp issues?

freedomtrain, 2020-02-23:
I fixed the problem by changing the kernel to 4.15.0 instead of 5.0.0. This [link](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) states that in order to use realsense on 5.0.0 you have to build from source which I have done but the mentioned problem still exist so I think I'll just stick with older kernel.

closed 2020-02-23 by freedomtrain