realsenseai/realsense-ros #1079: SR305 TX2 Jetpack4.2 align_depth value is error

weseu, 2020-02-12:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR305  | 
| Firmware Version                   |  librealsense 2.22.0 ; RealSense ROS v2.2.12 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9.140-tegra                                         | 
| Platform                                 | NVIDIA TX2  |

### Issue Description
the first, I fix the "/realsense-ros/realsense2_camera/cMakeLists.txt  . I chage "find_package" 2.29.0 to 2.22.0, and make it. 

runing:
roslaunch realsense2_camera rs_camera.launch align_depth:=true
and
rostopic echo /camera/aligned_depth_to_color/image_raw

and than, the massge that let me comfusion.  if I use D435i , and let object keep from near to far. I will get the following massage (it just a sample distance).

0 ,  100 --> 0 , 200 --> 1, 100 --> 2, 0   -------> the 16 bit massage will be bigger.

but if I use SR305. I will get the following massage.

0, 20 --> 0, 70 --> 0 , 127 --> 0 , 127 --> 0 , 100 --> 0 , 27  ----------> the massage is not be bigger,and the high byte has not any value.

please help me. thank you.

weseu, 2020-02-13:
By the way. My Jetpack version is 4.2.

and I try runing:
roslaunch realsense2_camera rs_camera.launch align_depth:=true
and
realsense-ros/realsense2_camera/scripts/show_center_depth.py

when I let it from near to far, I get the flowing massage for SR305.
/camera/depth/image_rect_raw: Depth at center(320,240): ..........
20 --> 70 --> 127 --> 127 --> 100 --> 50 --> 27 ----------> the massage is not be bigger

I get the same issue.

RealSenseCustomerSupport, 2020-02-14:
----------------------------------------------

The RealSense ROS version 2.2.11 and 2.2.12 have the supported libRealSense version 2.31.0. Please give 2.31.1 a try and see if the problem persists. Thank you.
https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.11

weseu, 2020-02-15:
this issue can follow https://github.com/IntelRealSense/librealsense/issues/5846 , so I close it

closed 2020-02-15 by weseu